mavlink/mavlink #658: Usage of "ID of message in payload" must be uint32_t compatible instead of uint8_t (v2.0)

samacumen, 2016-12-14:
Hello,

In the __mavlink_message struct (mavlink_types.h), msgid is defined to be 24 bit field (uint32_t msgid:24;).

However, the function mavlink_frame_char_buffer(...) passes the parameter "uint8_t c", which gets copied to msgid as follows:

```
MAVLINK_HELPER uint8_t mavlink_frame_char_buffer(mavlink_message_t* rxmsg, 
                                                 mavlink_status_t* status,
                                                 uint8_t c, 
                                                 mavlink_message_t* r_message, 
                                                 mavlink_status_t* r_mavlink_status)
{
..
..
 case MAVLINK_PARSE_STATE_GOT_COMPID:
 		rxmsg->msgid = c; 
..
}
```
Shouldn't the passed parameter be uint32_t as well? This limits the msgID defined in XML to 255. With new developments, more IDs may be required.

Regards,
Shyam

magicrub, 2016-12-14:
MAVLink <= v1.0 is 8bit, MAVLink v2.0 is 24bit.

samacumen, 2016-12-14:
@magicrub Agreed. But the problem I see, is here:
rxmsg->msgid = c; 

i.e. copying an 8 bit value to a 24 bit. So, that means msgID beyond 255 will not be supported/received at the receiving node.

samacumen, 2016-12-14:
Ok, get it. The following bits have been handled by MAVLINK_PARSE_STATE_GOT_MSGID1 and MAVLINK_PARSE_STATE_GOT_MSGID2 by << shifting to higher significant bits.

amilcarlucas, 2018-03-08:
I think it is better to close this one and open a new issue for the "user custom message ID" range.

closed 2018-03-08 by LorenzMeier