mavlink/mavlink #659: Wrong parameters are set in the message

SalahuddinAsh, 2016-12-22:
Hi everyone, 

I'm trying to send DO_MOUNT_CONTROL message from ODroid to PC. I use this snippet of code
```

mavlink_msg_command_long_pack(
				mavlink_system.sysid,    		// @param system_id 
				mavlink_system.compid,        		// @param component_id 
				&msg,                         		// @param msg The MAVLink message
				1,                  			// @param target_system System 
				0,               			// @param target_component Component 
				MAV_CMD_DO_MOUNT_CONTROL, 		// @param command Command ID
				0,                            		// @param confirmation
				4,                            	// @param param 1 - pitch 
				8,                               	// @param param 2 - roll 
				16,                            	// @param param 3 - yaw 
				0, 					// @param param 4 - WIP: alt in meters 
				0,                                  	// @param param 5 - WIP: latitude in degrees * 1E7
				0,                                  	// @param param 6 - WIP: longitude in degrees * 1E7
				MAV_MOUNT_MODE_MAVLINK_TARGETING);  	// @param param 7 - MAV_MOUNT_MODE enum value

  len = mavlink_msg_to_send_buffer(buf, &msg);
  serialWrite(fd, pbuf, len);
```

The problem is that ODroid sends a message with the wrong values to PC. Here is the sent message 
```
fe 21 00 14 c8 4c   00 00 80 40   00 00 00 41    00 00 80 41    00 00 00 00    00 00 00 00    00 00 00 00 00 00 00 40    cd 00 01 00 00 71   94
```
The angles' values (param1, param2, param3) are not set right in the MAVLink message and also param7 which is MOUNT_MODE which is MAVLINK_TARGETTING should be equal to 2 not 40 in hex.

Could anyone please tell me why does this happen so?
Many thanks

julianoes, 2016-12-24:
I don't think you can look at param1/2/3/7 in hex because these are floats. Once you round/cast them back on the PC side to int, they should be correct.

SalahuddinAsh, 2016-12-24:
@julianoes Man, I don't know how to thank you. I didn't pay attention to this point. After casting using the next code snippet I have the right values.
```
  char buffer[] = {0x00, 0x00, 0x80, 0x41};
  float f = *(float *)&buffer;
  printf("%f\n",f);
```
Million thanks

julianoes, 2016-12-24:
Cool, no worries!

closed 2016-12-24 by julianoes