mavlink/mavlink #665: generic estimator state and covariance

jgoppert, 2017-01-26:
The current local_position_ned_cov is ok for debugging but it assumes the states in the estimator. This makes it very brittle. Why don't we just make a generic state and covariance message, with size of about 25 states and the upper triangle of the covariance matrix?

jgoppert, 2017-01-26:
It would be cool if the estimator could also pass a state order, if we define this in mavlink as an enum somewhere?

For example

```bash
enum state_id {
  unused = 0,
  pos_N = 1,
  pos_E =2,
  pos_D = 3,
  asl = 4,
  agl = 5,
  baro_bias = 6,
  gryo_bias_x = 7
};
float state[25];
float variance[25];
uint8_t id[25];

id = {1, 2, 4, 5, 6, 7, ...., 0, 0, 0};
```

jgoppert, 2017-01-26:
@mhkabir @ecmnet @bkueng @LorenzMeier thoughts?

jgoppert, 2017-01-26:
I think after playing this we want 3 messages:

estimator_state {
  uint8_t n; // valid number of states
  float x[25];
  uint8_t id[25];
}

estimator_standard_deviation {
  uint8_t n;
  float std[25];
  uint8_t id[25];
}

estimator_covariance_matrix {
  uint8_t n;
  uint8_t cov[300];
  uint8_t id[25];
}

priseborough, 2017-01-27:
We need states, state variances, innovations and innovation variances to do proper debugging and tuning. I'm not fussed about how this gets packaged up provided there is tool support to parse it at the other end and the states and observations types used by the ecl ekf2 are supported.

jgoppert, 2017-01-27:
@priseborough do you see anything in MAV_FIELD that I didn't list and you need? I think since I added innovations, we need something for each measurement, so I plan on adding these.

Need to add:
* VISION: x, y, z, q(0-4)
* MAG: x, y, z
* LIDAR: dist_bottom
* SONAR: dist_bottom

amilcarlucas, 2018-03-08:
Now that there is a PR, can this be closed ?

hamishwillee, 2020-04-23:
Yes :-)

closed 2020-04-23 by hamishwillee