mavlink/mavlink #671: packet encoding in java library

b-rohit, 2017-02-13:
In mavlink java library encodePacket function make a final packet as byte[] and in C library it is done in uint8_t[]. so in java library encoded buffer has signed values. >
` public byte[] encodePacket() {
        byte[] buffer = new byte[6 + len + 2];
        
        int i = 0;
        buffer[i++] = (byte) MAVLINK_STX;
        buffer[i++] = (byte) len;
        buffer[i++] = (byte) seq;
        buffer[i++] = (byte) sysid;
        buffer[i++] = (byte) compid;
        buffer[i++] = (byte) msgid;

        final int payloadSize = payload.size();
        for (int j = 0; j < payloadSize; j++) {
            buffer[i++] = payload.payload.get(j);
        }
        generateCRC();
        buffer[i++] = (byte) (crc.getLSB());
        buffer[i++] = (byte) (crc.getMSB());
        return buffer;
    }` 

packet start sequence is 254 but when it copies to the buffer variable it becomes -2.  same happens other parameters >127.

WickedShell, 2017-02-13:
This isn't an issue. Java doesn't have an unsigned type, so you have to use a signed type. The values are identical, and the only use of the buffer is for building a packet to be flushed out, so you should never be peaking into it anyways.

This should be closed, but I don't have permissions on the MAVLink repo to close it.

b-rohit, 2017-02-14:
@WickedShell 
final encode packet in Java 
`void sendHeartBeat(){
msg_heartbeat msg = new msg_heartbeat();
  int counter=0;
  byte[] buffer;
  msg.custom_mode = 1;
  msg.type = 0;
  msg.autopilot = 0;
  msg.base_mode = 0;
  msg.mavlink_version = 1;
  MAVLinkPacket pck = msg.pack();
  buffer = pck.encodePacket();
 `
-2 9 0 -1 -56 0 1 0 0 0 0 0 0 0 1 -96 113
In C 
void sendHeartBeat(){
`mavlink_message_t msg;
 mavlink_system.sysid=255; 
 mavlink_system.compid=200; 
 uint8_t buff[MAVLINK_MAX_PACKET_LEN];
 int type = 0; // MAV_TYPE_QUADROTOR;  
 uint8_t autopilot_type = 0; 
 uint8_t base_mode = 0; 
 uint32_t custom_mode = 1;     
 uint8_t system_state = MAV_STATE_STANDBY;
 mavlink_msg_heartbeat_pack(mavlink_system.sysid, mavlink_system.compid, &msg , type , autopilot_type , base_mode , custom_mode , system_state );
 uint16_t len=mavlink_msg_to_send_buffer(buff, &msg);`
}

254 9 218 255 200 0 1 0 0 0 0 0 0 3 3 165 172

Problem 2. 
Packet Sequence is not updated in Java.
`public byte[] encodePacket() {
        byte[] buffer = new byte[6 + len + 2];

        int i = 0;
        buffer[i++] = (byte) MAVLINK_STX;
        buffer[i++] = (byte) len;
        buffer[i++] = (byte) seq;
...`
seq value is always zero.

In C library it is incremented by one.
`MAVLINK_HELPER uint16_t mavlink_finalize_message_chan(mavlink_message_t* msg, uint8_t system_id, uint8_t component_id,
                                                      uint8_t chan, uint8_t length)
#endif
{
        // This code part is the same for all messages;
        uint16_t checksum;
        msg->magic = MAVLINK_STX;
        msg->len = length;
        msg->sysid = system_id;
        msg->compid = component_id;
        // One sequence number per component
        msg->seq = mavlink_get_channel_status(chan)->current_tx_seq;
        mavlink_get_channel_status(chan)->current_tx_seq = mavlink_get_channel_status(chan)->current_tx_seq+1;
`

WickedShell, 2017-02-14:
@b-rohit you manage the seq num yourself with the Java bindings. There are no concepts of channels that it manages for you either, the Java bindings only hand you support for generating packets as needed. To have a seq num simply set it on your outgoing packet.

I don't see what you are getting at with the strings you pasted, the two issues provided different settings to the packets so of course they encoded differently.

b-rohit, 2017-02-16:
@WickedShell thanks for help I have implemented your suggestion. it works for me now. I suggest a small change for sequence number in packet. Increase the variable `seq` in `public byte[] encodePacket()` function.
`public byte[] encodePacket() {
...
...
seq++;
return buffer;
 }`

closed 2017-02-16 by b-rohit