mavlink/mavlink #680: mavgenerate.py future module not found!!

badaniels, 2017-03-15:
Hi,

I'm trying to add a mavlink msg and also change a commons message. I'm now trying to convert the xml file on my mac but whenever I try and run the mavgenerate.py file it says that a future module was not found. I've downloaded future-0.16.0 for python but it stills say the same. 

WTF is happening or going wrong. Why can't I just write or amend the .h files for mavlink messages. 

Thanks in advance.

cclauss, 2017-03-16:
```
$ python -V
$ pip -V
$ pip install --upgrade future  # http://python-future.org/quickstart.html
```

badaniels, 2017-03-16:
Thanks. I'll try that this afternoon. Is it possible to just write a .h file, or do you have to write the xml and then convert it?

closed 2017-03-16 by badaniels