mavlink/mavlink #682: CRC_EXTRA = 0 when combining multiple message definitions for mavgenerator

Neoster, 2017-03-17:
Hi All,

Recently I had a functioning arduino sketch talking MAVLink1 over RF.
Now i decided to update to MAVLink2 and include the common.xml set of messages.

What i did is create my own message definition on top of the common.xml and pixhawk.xml
I assigned my message definition the name shift.xml
This is what the beginning on my file looks like:

`<?xml version='1.0'?>
<mavlink>
    <include>common.xml</include>
    <include>pixhawk.xml</include>
    <messages>
        
        
        <message id="500" name="MAV_SHIFT_RADIODOG_IDENTIFICATION_MSG">
`

The sketch compiles and runs. I can across the parse_state MAVLINK_PARSE_STATE_GOT_BAD_CRC1.
I decided to log the complete crc calculation on both the encode as the decode part of the code. Now it seems that the encode side knows about the CRC_EXTRA value for my messageId 500 which is 203 (0xCB). However the decoding side of the code does not know about this CRC_EXTRA value.

In the file "mavlink_helpers.h" is the function
 `MAVLINK_HELPER` const mavlink_msg_entry_t *mavlink_get_msg_entry(uint32_t `msgid)`
In here i did log all the known elements of mavlink_message_crcs.
`static const mavlink_msg_entry_t mavlink_message_crcs[] = MAVLINK_MESSAGE_CRCS;`
Since MAVLINK_MESSAGE_CRCS is a define located in all my sub-groups of MAVLink {common/common.h, pixhawk/pixhawk.h, shift/shift.h} only one, being the first, is the define used. Therefore the decoding part of MAVLink is unable to find the CRC_EXTRA value for all messages NOT in sub-group common.

To prove I am missing the CRC_EXTRA value, i did change this line:
`uint8_t crc_extra = e?e->crc_extra:0;`
to
`uint8_t crc_extra = e?e->crc_extra:203;`
resulting in a succesfull decoded message.

magicrub, 2017-03-17:
Can you post a link to your repo+branch so we can see it? I have a feeling there's something else going on

Neoster, 2017-03-17:
Sure, if you can tell me how.
Sorry for my lack of knowledge on this part of GIT.
I did fork the mavlink repo yesterday, brach 'master'

magicrub, 2017-03-17:
make new branch, commit+push all your source to it, post a link to the github repo

Neoster, 2017-03-17:
Will do

Neoster, 2017-03-17:
Sorry for the holdup.
The repo we are working on is a private repo since it contains a lot of other code related to our project.
I tried making a new branch in the mavlink repo and publish it, but it seems I am not allowed to do so.

magicrub, 2017-03-17:
well, then I/we are of limited help to you.

closed 2018-02-21 by LorenzMeier