mavlink/mavlink #689: win7 x64 sp1, python 2.7.3,can't generate the mavlink code

wmx-github, 2017-03-28:
win7 x64 sp1, python 2.7.3,can't generate the mavlink code, 
it runs quicklly  and nothing happend

wmx-github, 2017-03-28:
File "pymavlink\generator\mavgen.py", line 12,       (  from future import standard_library  )

magicrub, 2017-03-28:
"ImportError: No module named future"
You need to install future. We're hearing a lot of people having trouble with this so we should look into the installation tool and see why it's not installing it correctly..

amilcarlucas, 2018-02-19:
Not a bug.

IMHO this can be closed now.

hamishwillee, 2018-02-20:
Future is not installed automatically. This PR will fix the docs  instructions to mention the requirement: https://github.com/mavlink/mavlink-devguide/pull/40

closed 2018-02-20 by hamishwillee