mavlink/mavlink #693: MavLink 2 C headers broken

DonLakeFlyer, 2017-04-01:
The api for mavlink_msg_mission_count_pack_chan now has an extra function param for MAV_MISSION_TYPE. But the api for mavlink_msg_mission_count_encode does not. They should both have the new function argument shouldn't they?

vooon, 2017-04-02:
```c
/**
 * @brief Encode a mission_count struct
 *
 * @param system_id ID of this system
 * @param component_id ID of this component (e.g. 200 for IMU)
 * @param msg The MAVLink message to compress the data into
 * @param mission_count C-struct to read the message contents from
 */
static inline uint16_t mavlink_msg_mission_count_encode(uint8_t system_id, uint8_t component_id, mavlink_message_t* msg, const mavlink_mission_count_t* mission_count)
{
    return mavlink_msg_mission_count_pack(system_id, component_id, msg, mission_count->target_system, mission_count->target_component, mission_count->count, mission_count->mission_type);
}
```

Just generated v2 and i do not see any issue.

DonLakeFlyer, 2017-04-03:
My confusion. This is fine.

closed 2017-04-03 by DonLakeFlyer