mavlink/mavlink #697: MAV_BATTERY_TYPE/MAV_BATTERY_FUNCTION enum collision

WickedShell, 2017-04-07:
There is a ```MAV_BATTERY_TYPE_PAYLOAD``` stuck inside of the ```MAV_BATTERY_FUNCTION``` enum https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2194 this should be ```MAV_BATTERY_FUNCTION_PAYLOAD```.

Is anyone using this/or would experience breakage if we fixed the name on the payload one to be ```MAV_BATTERY_FUNCTION_PAYLOAD```

This raises a generator question: If any param in the group isn't prefixed with the group name that should result in an error?

DonLakeFlyer, 2017-04-07:
Yeah looks like a typo. The naming rule makes sense to me as well.

LorenzMeier, 2018-02-21:
@WickedShell Could you send a PR fixing this?

hamishwillee, 2026-08-25:
Fixed many years ago

closed 2026-08-25 by hamishwillee