mavlink/mavlink #723: Multiple MAVLinks

AliSiddiqui4, 2017-05-09:
Hi, 

I have managed to create multiple instances of MAVlinks they have different udp and tcp. But I can't get their udp bind address to be different. They have to be unique so I could easily control multiple erle copters in Gazebo.

magicrub, 2017-05-09:
what is the scope of your question? Is this a problem with an app you wrote or with mavproxy?

AliSiddiqui4, 2017-05-10:
I am trying simulate multiple Quadcopter in Gazebo. I have followed the ErleCopter tutorials form here: [tutorials](http://docs.erlerobotics.com/simulation/intro). But the tutorials are only for a single quadcopter, and in order for me to simulate multiple quads I need multiple instances of MAVlink with different port addresses etc.

magicrub, 2017-05-10:
Here's another resource:
http://ardupilot.org/dev/docs/using-gazebo-simulator-with-sitl.html

I think what you need is to run multiple instances of sim_vehicle.py, that allows for multiple IP ports. Each instance uses a port number 1 higher. I can't remember the argument you add to sim_vehicle though

magicrub, 2017-05-10:
maybe it's -I (that's an eye). And maybe the port is 10 higher instead of 1. Give that a try and report back!

AliSiddiqui4, 2017-05-11:
@magicrub I did do that you change the Instances argument and the port change by 10. All the ports change by 10 except the udp bind port it remains the same.

magicrub, 2017-05-11:
sounds like your issue is posted in the wrong spot. This is not a "mavlink" problem but a Gazebo driver interface problem.  Please repost on the Ardupilot/ardupilot repo

amilcarlucas, 2018-02-21:
IMHO this can be closed here

closed 2018-02-21 by LorenzMeier