mavlink/mavlink #743: [c-library] encode message without modifying  global state

MikeGitb, 2017-06-27:
Correct me if I'm wrong, but it seems to me as if the c library doesn't provide any way to pack the data of a mavlink message into a `mavlink_message_t` without accessing some global data (namely one of the predefined channels).
This can be problematic in a multithreaded scenario and/or if the library is used in different parts of the program and the author doesn't make sure they are using different channels.

Have you considered extending the API, such that a user can simply create and pass a local channel variable to the respective functions

vooon, 2017-06-28:
C-library has `mavlink_finalize_message_buffer()` and `mavlink_frame_char_buffer()` which independent of global mavlink_state_t array (you should provide pointer).
But there is no C pack functions, this `_buffer()` functions used by C++11 gen (to hide state inside connection class).

MikeGitb, 2017-06-28:
@vooon: There is a c++ wrapper?

TSC21, 2017-06-28:
@MikeGitb https://github.com/mavlink/mavlink/commit/dc415938cea1cde5d5e2eebbb27a920e67c1ec4e

vooon, 2017-06-28:
@MikeGitb it's not fully wrapper, only uses helpers from C-lib. Message serialization done in C++ classes.

MikeGitb, 2017-06-28:
Thanks for the explanation. I'll have a look and see if I can use it.

vooon, 2017-06-28:
You may use [libmavconn](https://github.com/mavlink/mavros/tree/master/libmavconn) as a reference.
But i still not done v2 packet signing feature...

amilcarlucas, 2018-02-20:
@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here.

hamishwillee, 2018-02-20:
@MikeGitb can you make that report?

amilcarlucas, 2018-02-21:
@hamishwillee does it make any sense to keep these things open ?
Why don't you close this. This list of issues is very big, and I do not see any developer taking a look at it and fixing things. Making the list smaller and cleaner might help developers to take action and actually FIX issues.
Support questions belong in the forum. PX4, pymavlink, Ardupilot, QGC,... issues belong on their own repositories. If you do not want to close stuff, give me permissions, I'll close it for you !

LorenzMeier, 2018-02-21:
Yeah, I agree that should go.

closed 2018-02-21 by LorenzMeier

hamishwillee, 2018-02-21:
@LorenzMeier Perhaps you could give @amilcarlucas access to close these issues? 

@amilcarlucas Not really - I'm just cautious closing issues because I don't have responsibility for MAVLink or a clear policy to work against (I only have "rights" because I'm handling some documentation and tooling work). I've closed where obvious mostly because I didn't want your review work to go to waste.