mavlink/mavlink #756: Use of centiAmps - suggest Amps or milliAmps

jliphard, 2017-07-15:
Right now, the battery current is an int16 and has units of centiAmps (100 centiAmps = 1 Amp). This is a problem for two reasons: (1) it's confusing, and (2) it prevents the future use of precision current sensors and current monitoring which will have mA resolution. The limitation could be resolved by moving to a float, and using units of either A or mA, or using int32, and using mA. 
```
<field type="int16_t" name="current_battery" units="cA">Battery current, in 10*milliamperes (1 = 10 milliampere), -1: autopilot does not measure the current</field>
```
To maintain backwards compatibility, a new message could be added:
```
<field type="int32_t" name="current_battery_mA" units="mA">Battery current in milliAmperes, -1: autopilot does not measure the current</field>
```

amilcarlucas, 2018-01-22:
(1) is addressed by adding a documented list of known units: https://github.com/mavlink/pymavlink/pull/5/commits/5f33f6fee9af3141c0293fb740f8ef95c0f11291

(2) Adding extra four bytes to get more precision is probably never going to happen.

amilcarlucas, 2018-02-19:
The flight stack is responsible for doing fast measurements and data integration. It makes no sense to send high resolution data over the air at low refresh rates.

IMHO this can be closed now.

hamishwillee, 2018-02-20:
@jliphard I'll close this now. Please reopen with a response if you think this requires further thought. Thanks @amilcarlucas

closed 2018-02-20 by hamishwillee