mavlink/mavlink #757: mavlink_sha256.h: mavlink_sha256_ctx wrong type size on gccavr

az0uz, 2017-07-19:
when compiling the C version, I get the warning:
```
mavlink_sha256.h: In function 'void mavlink_sha256_final_48(mavlink_sha256_ctx*, uint8_t*)':
mavlink_sha256.h:212:36: warning: right shift count >= width of type
zeros[dstart+5] = (m->sz[0] >> 16) & 0xff;
^
mavlink_sha256.h:213:36: warning: right shift count >= width of type
zeros[dstart+4] = (m->sz[0] >> 24) & 0xff;
^
mavlink_sha256.h:216:36: warning: right shift count >= width of type
zeros[dstart+1] = (m->sz[1] >> 16) & 0xff;
^
mavlink_sha256.h:217:36: warning: right shift count >= width of type
zeros[dstart+0] = (m->sz[1] >> 24) & 0xff;
^
```

I think to fix it, this:

julianoes, 2017-07-22:
@az0uz thanks for reporting.
@tridge is there a reason that `unsigned int` is used?

az0uz, 2018-01-16:
Can we merge ?

magicrub, 2018-01-16:
I've pinged @tridge about it.

magicrub, 2018-01-16:
mavlink/mavlink will need a submodule update from ardupilot/pymavlink for it to take effect here

julianoes, 2018-01-17:
@magicrub thanks for the note.

amilcarlucas, 2018-01-22:
mavlink/mavlink#826 fixes this, and passes CI tests. I hope it gets merged soon.

closed 2018-01-22 by az0uz