mavlink/mavlink #766: VIDEO_STREAM_INFORMATION lacks video Aspect Ratio

dogmaphobic, 2017-08-01:
Is anybody already using this? It's marked as "WIP".

Can I add an aspect_ratio field to it or should I create a new pair of command/response just for it?

My idea is to just add this to the end of the message:

```
      <field type="float" name="aspect_ratio">Video image aspect ratio</field>
```

hamishwillee, 2020-04-23:
@dogmaphobic Bringing this back to your attention. I have no idea who is using this, but probably only QGC.

Perhaps the best way to test this is to just create a PR if you think this would be useful. Otherwise can we close this?

dogmaphobic, 2020-04-23:
Nah, this would only be an issue for streams with non square pixels (anamorphic). I have yet to see one. In 99.9% of the cases, you can derive it from its frame size.

closed 2020-04-23 by dogmaphobic

ben-crowhurst, 2020-08-09:
Hey, I was looking to implement VIDEO_STREAM_INFORMATION as a way of getting available video feeds.  As @dogmaphobic noted in 2017 it was, and still is, marked as WIP.

Will this message ever get ratified and make it into the mainstream protocol?

ben-crowhurst, 2020-08-09:
Sorry just noted I've hijacked a thread, I'll create separate issue.