mavlink/mavlink #769: Update Web Reference Documentation w/ Extension Markers

BenArtes, 2017-08-10:
The generated online documentation doesn't include notes or references about extensions.  

I think it would help if the generated documentation had more references to the differences caused by MAVLink 2 like extensions.

I've made some changes as a suggestion in my Fork: https://github.com/BenArtes/mavlink/commit/b0693a3cb886b5fc397ef6e7efa942ec08ed76aa.

>Note: I DO NOT know XSLT,  I spent the morning hacking this together to demonstrate the idea and potentially contribute it, but it needs to be cleaned up.

End Result:
[image]

If this is helpful I can try to clean it up and submit a PR.

Two Asides:

* Not sure what else could be added to clarify things.  Any thoughts?
* There are a number of grammatical/spelling mistakes on the mavlink.org pages.  Are those pages hosted on GitHub anywhere?

TSC21, 2017-08-10:
I like the idea!
> Not sure what else could be added to clarify things. Any thoughts?

What about changing the `Field Name` color instead of adding a new line with `EXTENSIONS` description?

BenArtes, 2017-08-10:
I'd be open to that, but I wouldn't know where to begin implementing that in XSLT.  Adding an extra row when an `<extensions/>` tag was easy enough but I don't know enough about XSLT to know if or how easily it supports changing the response to a tag based on tags that came before it.

TSC21, 2017-08-10:
```xsl
table.sortable td.mavlink_name_ext {
     color:#9FEF81;
     font-weight: bold;
     width: 25%;
     vertical-align: top;
 }
```
Then, what if we create a new table when `test="not(@name)"` and apply the above CSS class to `@name` values?