mavlink/mavlink #771: Cant parse Mavlink packet

Aurusd, 2017-08-18:
Have intel aero drone with flight controller (FC). Get byte array received from FC:
-2, -55, 118, 66, 0, 0, 0, -128, 30, 1, -91, 117, -3, 14, 0, 0, 37, 1, 1, 111, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -48, 24, -91, 112, -84, 4, 106, -33, -3, 40, 0, 0, 38, 1, 1, -116, 0, 0, 120, -32, 73, 50, 1, 0, 0, 0, -28, 29, -100, 59, -92, -22, 46, -68, -91, -120, -42, 62, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, -128, -65, 7, -112, -3, 28, 0, 0, 39, 1, 1, 30, 0, 0, 2, 105, 78, 0, 88, -8, -120, -69, -4, -74, 45, 60, 61, 68, -93, -65, 10, 28, -24, -70, 79, -85, -69, -69, 46, 69, -2, 57, -17, -114, -3, 32, 0, 0, 40, 1, 1, 31, 0, 0, 29, 105, 78, 0, -16, -84, 77, 63, 79, -106, -65, 58, 38, -4, -77, 59, 48, 107, 24, -65, 40, -61, -90, -70, 36, 46, 88, -69, -61, -83, 59, -69, 60, -63, -3, 28, 0, 0, 41, 1, 1, 30, 0, 0, 41, 105, 78, 0, -17, -91, -119, -69, 82, -109, 44, 60, 5, 69, -93, -65, 123, 62, -93, -70, -90, -95, 106, -69, 60, 20, 91, -71, 11, 118, -3, 1, 0, 0, 42, 1, 1, 42, 0, 0, 0, -77, 88, -3, 62, 0, 0, 43, 1, 1, 105, 0, 0, -31, -39, 74, 50, 1, 0, 0, 0, 23, -91, 103, 61, 60, 69, -115, 60, 71, -62, 28, -63, 60, 42, 22, -69, -27, -49, -117, -69, -71, -53, -59, -70, 81, -5, -89, 61, 121, -43, -119, 62, 54, 60, -110, 61, 0, 0, 0, 0, 0, 0, 0, 0, 38, 119, 120, 66, 20, -82, 41, 66, -1, 27, -99, -76, -3, 28, 0, 0, 44, 1, 1, 32, 0, 0, 61, 105, 78, 0, 0, 0, 0, 0, 0, 0, 0, 0, -67, 113, -104, -64, -76, 18, -104, -71, -45, 30, 81, 59, -10, 72, -44, -68, -34, 68, -3, 28, 0, 0, 45, 1, 1, 30, 0, 0, 77, 105, 78, 0, 69, -17, -120, -69, 117, -2
and i cant detect here a packet, for mavlink 2.0 Header is 0xFE?

magicrub, 2017-08-18:
PROTIP: dec(-2) == hex(0xFE)

Treat it as uint8_t hex stream and it should make more sense. The secret to MAVlink is you shouldn't even be looking at this, all the MAVLink parsing helper functions handle this for you.

WickedShell, 2017-08-18:
MAVLink2 has a magic of 0xFD, 0xFE is MAVLink1

LorenzMeier, 2017-08-18:
I would recommend the dev guide at http://mavlink.io and the C library: https://github.com/mavlink/c_library_v2

- or any other language binding you might choose.

As @magicrub said: There is no point in looking at this

closed 2017-08-18 by LorenzMeier

Aurusd, 2017-08-19:
MAVLink2 has a magic of 0xFD, 0xFE is MAVLink1
is a right answer, thanks WickedShell. But there is a question why i cant find this information on mavlink official docs.