mavlink/mavlink #780: Attribute error, generator fails.

vvranjek, 2017-09-21:
When I try to generate common.xml mavlink libs (C, v2.0), generator gives a following error:
```
Errors occurred in mavgen:
Element 'field', attribute 'display':
The attribute 'display' is not allowed., line 2684
```
If I remove all the 'display' attributes, it starts complaining about all the unit attributes, such as m, s, m/s, etc..
After the removal of all unsupported attributes it generates successfully.

A bit off-topic: With which version of mavlink generator are mavlink libs generater for PX4 v1.6.5?

magicrub, 2017-09-21:
Try this one: https://github.com/ArduPilot/pymavlink

amilcarlucas, 2017-09-21:
well the real fix is this one https://github.com/ArduPilot/pymavlink/pull/118

vvranjek, 2017-09-22:
Thanks, but I'm confused, is https://github.com/mavlink/mavlink/ obsolete? Why are you pointing me to an ArduPilot repo?  Is PX4 using ArduPilot's mavlink generator?

magicrub, 2017-09-22:
The generator is part of Ardupilot. Mavlink links to it as a submodule. The ArduPilot version is the most up-to-date.

LorenzMeier, 2017-09-22:
@vvranjek The MAVLink generator used for the standard / Dronecode is located here and is the canonical standard. For anything that is Dronecode / standard please use the repository here.

The specific issue you're having seems to be related to generating common.xml instead of standard.xml, which is what's being used to generate the standard library: https://github.com/mavlink/c_library_v2

Try that instead.

closed 2017-09-22 by LorenzMeier

joshlevin88, 2017-11-24:
Generating standard.xml produces the same effect.