mavlink/mavlink #781: Missing MAV_CMD's parameters names and optional enum

superware, 2017-09-24:
Hello,

```
<entry value="176" name="MAV_CMD_DO_SET_MODE">
	<description>Set system mode.</description>
	<param index="1">Mode, as defined by ENUM MAV_MODE</param>
	<param index="2">Custom mode - this is system specific, please refer to the individual autopilot specifications for details.</param>
	<param index="3">Custom sub mode - this is system specific, please refer to the individual autopilot specifications for details.</param>
	<param index="4">Empty</param>
	<param index="5">Empty</param>
	<param index="6">Empty</param>
	<param index="7">Empty</param>
</entry>
```

Should actually be:

```
<entry value="176" name="MAV_CMD_DO_SET_MODE">
	<description>Set system mode.</description>
	<param index="1" name="mode" enum="MAV_MODE">Mode, as defined by ENUM MAV_MODE</param>
	<param index="2" name="custom_mode">Custom mode - this is system specific, please refer to the individual autopilot specifications for details.</param>
	<param index="3" name="custom_sub_mode">Custom sub mode - this is system specific, please refer to the individual autopilot specifications for details.</param>
	<param index="4">Empty</param>
	<param index="5">Empty</param>
	<param index="6">Empty</param>
	<param index="7">Empty</param>
</entry>
```

So that it's possible to generate corresponding MAV_CMD classes. If one of the main reasons for this XML is code generation, then parameters names and enums are simply missing.

Can I arrange a PR?

Thanks!

amilcarlucas, 2018-01-22:
This one already exists in ArduPilot/pymavlink#7
 maybe you could work on that one?

amilcarlucas, 2018-03-08:
This one can IMHO be closed.

amilcarlucas, 2019-04-05:
Implemented in master now.

closed 2019-04-05 by amilcarlucas