mavlink/mavlink #783: False generation of mavlink headers (with bad XML input)

samacumen, 2017-10-04:
I noticed incorrect .h header file generation while using mavgenerate.py with my custom XML file. The message when looks like the following generates incomplete/incorrect .h file 

> <message id="500" name="TEST_MESSAGE">
> 	<field type="uint16_t" name="data1">Some data </field>
> 	<field type="uint16_t" name=""></field>
> 	<field type="uint16_t" name="data3"></field>
>     </message> 

Please note the name for the 2nd field above is missing. The struct generated is:

> MAVPACKED(
> typedef struct __mavlink_test_message_t {
>  uint16_t data1; /*< The number of packets received during */
>  **uint16_t ; /*< */**
>  uint16_t data3; /*< */
> }) mavlink_test_message_t;

Intuitively, the user would assume the mavgenerate.py to throw an error. It took me a while to spot that the compilation was complaining about this silly error when the compiler was pointing at something entirely different.

amilcarlucas, 2018-01-22:
This is a pymavlink issue. Please report it upstream at https://github.com/ardupilot/pymavlink/issues

amilcarlucas, 2018-02-20:
@samacumen please do as I told you or it wont get fixed!

@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here.

samacumen, 2018-02-20:
Thanks, reported at pymavlink [here](https://github.com/ArduPilot/pymavlink/issues/150)

hamishwillee, 2018-02-20:
Thanks guys. Closed.

closed 2018-02-20 by hamishwillee