mavlink/mavlink #787: Building mavlink-gbp-release fails

fjulian, 2017-10-17:
I would have posted this into the [mavlink-gbp-release](https://github.com/mavlink/mavlink-gbp-release) repo but there is no issue section there.

When trying to install mavlink as 3rd party library for ROS, building it from source fails. Running the command

`sudo ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --install-space /opt/ros/kinetic`

builds and installs all necessary packages. When it reaches mavlink, two errors occur:

`CMake Error: File /home/pi/catkin_ws/src/mavlink/patch/config.h.in does not exist.`
and
`CMake Error: File /home/pi/catkin_ws/src/mavlink/patch/COPYING does not exist.`

The full console output for the mavlink part is contained in the following file:
[Mavlink_build_error.txt](https://github.com/mavlink/mavlink/files/1392953/Mavlink_build_error.txt)

The output says that certain files specified in the CMakeList.txt are missing, which is the case indeed. The specified files are also missing when browsing the [repo](https://github.com/mavlink/mavlink-gbp-release/tree/master/patch) online.

Is this an issue with the CMakeList.txt or am I doing something wrong?

Before running the build command, I followed the instructions on [how to install mavros/mavlink](https://dev.px4.io/en/ros/mavros_installation.html) and on [how to install ROS packages in general](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi) (see section 4.2).

If that helps: the output of `uname -r` is `Linux - 4.9.35-v7+ - armv7l`.

Any help is greatly appreciated.

vooon, 2017-10-20:
Issue tracker disabled because it is mostly for automatic distribution, rather that for any other.
And any errors with that repo is more related to mavros, than mavlink itself.

What branch you trying to build? And check https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation

fjulian, 2017-10-26:
Thanks for the reaction!

Regarding the branch I'm trying to build: I ran the command `rosinstall_generator mavlink | tee -a /tmp/mavros.rosinstall` which should get the latest released mavlink package.

I tried following the instructions from the link you posted but it doesn't work either. As I installed ROS from source, followed exactly the instruction on [this page](https://github.com/mavlink/mavros/blob/master/mavros/README.md#installation) under _Source Installation_ but when running the command `rosdep install --from-paths src --ignore-src --rosdistro kinetic -r --os=debian:jessie`, most of the packages are not found:
`ERROR: the following rosdeps failed to install`
  `apt: command [sudo -H apt-get install ros-kinetic-nav-msgs] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-visualization-msgs] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-tf] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-urdf] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-tf2-eigen] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-diagnostic-updater] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-control-toolbox] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-sensor-msgs] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-geographic-msgs] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-eigen-conversions] failed`
  `apt: command [sudo -H apt-get install ros-kinetic-tf2-ros] failed`

The detailed output for each of these commands is:
`executing command [sudo -H apt-get install ros-kinetic-<...>]`
`Reading package lists... Done`
`Building dependency tree`       
`Reading state information... Done`
`E: Unable to locate package ros-kinetic-<...>`

Googling this issue brought me to [this question and answer](https://answers.ros.org/question/250738/cannot-download-ros-kinetic-ros-tutorials/), but even after ensuring that the ROS repository is up-to-date, the errors persisted.

@vooon do you have any other idea what could be the problem?

vooon, 2017-10-26:
If you tries install to raspbian, then it may require to install whole system from source.
`rosinstall_generator` can find all dependant packages, note `--tar` may be faster.

Another option is to use ubuntu.

fjulian, 2017-10-26:
I will try to switch to Ubuntu. In [this post](https://github.com/mavlink/mavros/issues/709) someone had the same issue and it worked for them on Ubuntu Mate 16.04. This is what I will try to do.

@vooon thanks for your help.

closed 2017-10-26 by fjulian