mavlink/mavlink #788: MAV_CMD Waypoint with local position data

aykutkabaoglu, 2017-10-19:
Hi,
I would like to sent waypoint message to PX4 with local position data. There is a communication between pixhawk and computer via MAVROS. MAV_CMD_NAV_WAYPOINT is suitable to send waypoint data with geographic coordinate system and there are some MAV_CMD commands which include local position data such as MAV_CMD_NAV_LAND_LOCAL and MAV_CMD_NAV_TAKEOFF_LOCAL but i couldn't find a way to send waypoint information in metric format. How can i use MISSION_ITEM ( #39 ), MISSION_ITEM_INT ( #73 ) or SET_POSITION_TARGET_LOCAL_NED ( #84 ) messages to send local position information to PX4?
Thanks

TSC21, 2017-10-19:
https://dev.px4.io/en/ros/offboard_control.html

aykutkabaoglu, 2017-10-19:
By using rosservice /mavros/mission/push, the user is able to send waypoint without need to change the flight mode to offboard and; in this service message, i can use local positions for takeoff and land with only changing MAV_CMD number if i get it correctly. Therefore, isn't it possible to send waypoint with local position data or is it the only choice that publish geometry_msgs at /mavros/setpoint_position/local topic in offboard mode?

TSC21, 2017-10-20:
@aykutkabaoglu you can also use DroneCore instead of MAVROS, but for local position control ATM you need to use Offboard mode, as mission items regard on global coordinates.

aykutkabaoglu, 2017-10-21:
thank you,
what do you think to publish missions as global coordinates for local positioning? If i transform metric positions to geodetic coordinates for a specific region, what will be the sensitivity of global coordinates, mathematically?

TSC21, 2017-10-21:
@aykutkabaoglu check what is done on [`fake_gps`](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/fake_gps.cpp) plugin from MAVROS. Maybe it fits your needs. It basically converts from local to global coordinates and sends them as GPS data that is used by the flight controller. It's not used by the position controller but rather by the position estimator, so if you have a mission (global coordinates) it will estimate is position based on this fake GPS data. But you need a source of local position so this can be done, so I assume you are using some sort of vision system (like visual odometry, optical flow, motion capture).

aykutkabaoglu, 2017-10-23:
I use beacons to publish fake gps data to pixhawk by MarvelMind and, mavros px4 /tf transform is used by enabling the 'tf send' parameter in px4_config.yaml file. In short, pixhawk can get gps coordinates but the connected satellite number is 0 and it can be a problem for sending /tf data. If i can obtain correct tf information from mavros, i can use both local and global positions. During getting tf information from mavros, i can only get 0 for all coordinates by global_position sending to tf. 
I only changed the 'send' parameter in this part and probably, i need to describe more to obtain meaningful /tf data.
```
global_position:
  frame_id: "map"           # pose and fix frame_id
  rot_covariance: 99999.0   # covariance for attitude?
  use_relative_alt: true    # use relative altitude for local coordinates
  tf:
    send: true               # send TF?
    frame_id: "map"  # TF frame_id
    global_frame_id: "earth"  # TF earth frame_id
    child_frame_id: "base_link" # TF child_frame_id
```

TSC21, 2017-10-23:
@aykutkabaoglu I'm not quiet understanding what you are trying to do. `global_position` plugin is used to **retrieve** global position data from the FCU, not send. For sending `global_position` data as fake GPS coordinates, and as I said in my previous comment, you have to use [`fake_gps`](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/fake_gps.cpp) plugin.

TSC21, 2017-10-23:
Also, this is not a discussion to have here, as this is MAVROS related.

aykutkabaoglu, 2017-10-23:
Actually, i was trying to retrieve global position from the flight controller but when i read /tf informations all coordinates are 0, tf didn't include GPS data. I will take a look to both [fake_gps](https://github.com/mavlink/mavros/blob/master/mavros_extras/src/plugins/fake_gps.cpp) and [global_position](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/global_position.cpp) in detail and you are right this topic is related MAVROS anymore. Thank you

amilcarlucas, 2018-03-08:
This can IMHO be closed

closed 2018-03-08 by LorenzMeier