mavlink/mavlink #789: c# (csharp) mavlink Serialize issue

jmta, 2017-10-24:
Hi All,
I'm trying to send some packets with the c# wrapper, I've been receiving them fine but am having issues with sending currently. I am using the mavlink message Msg_set_mode using the following code snippet.

`MSg_set_mode mode = new Msg_set_mode();
mode.target_system = 1;
mode.base_mode = 209; //These Match the setting I've been using with c++ wrapper
mode.custome_mode = 2;
byte[] send = null;
send = new byte[11];
int offset = 0;
mode.Serialize(send, ref offset);
string  SendString = "";
foreach(byte byteval in send){
 SendString = SendString + byteval.tostring();
}
_serialPort.WriteLine(SendString);
Console.WriteLine(SendString); //Just for debug`

The output from the final line being "2000120900000"

Am I missing a step here to how to use the function correctly? Any help would be appreciated.

amilcarlucas, 2018-02-19:
The code has changed a lot, please retest with the latest code from master.

hamishwillee, 2018-02-20:
That may be difficult - CS building broken: https://github.com/mavlink/mavlink/issues/482

amilcarlucas, 2018-02-20:
@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here.

hamishwillee, 2018-02-20:
@jmta Can you please report upstream.

amilcarlucas, 2018-03-08:
Now that it is reported upstream, can it be closed here ?

closed 2018-03-08 by LorenzMeier