mavlink/mavlink #794: missing memoranda from v2_playground

wjwwood, 2017-11-10:
This link https://github.com/mavlink/v2_playground/issues/1, is referenced from this document: https://docs.google.com/document/d/1HDOfb3cQ6WiYm9M02RjAgyXRRNhMlGJF1iaLBAVp-T8/edit

I believe it contained a table comparing the encoding size of a few different options considered. I was hoping to point someone at it when I found the repository has been deleted. Does anyone on the mavlink team have some of that content floating around in their emails or something? I'd love to still have that information somewhere.

vooon, 2017-11-10:
Checked my mail, unfortunately my comment not saved. And i do not have a backup.

As i remember relations:
1. mavlink v1 = size of packed struct, == CDR, == LCM. (~56 bytes)
2. protobuf v2 near to p1 but at least +1 byte for each field, nested type add at least 1 byte
3. capt'n'proto similar to p2, but subtypes takes +8 bytes, with compression ~p2 or a bit better
4. CBOR, == MsgPack. Schema-less, smaller than compact JSON, but not much (~120 bytes)
5. Comact JSON (~150 bytes)
6. Pretty JSON (~350 bytes)

wjwwood, 2017-11-10:
Cool thanks for the recollection, I'll close this for now, but if anyone has the exact table in their email or something I'd still love to see it.

closed 2017-11-10 by wjwwood