mavlink/mavlink #795: Protocol definition XML format structural enhancement

kenkeiter, 2017-11-14:
Hi folks –
Given the random/broken state of the Go language implementations of MAVLink, I began work on a new one that would "catch up" (so to speak) with the current state of the protocol, do things somewhat more idiomatically, provide better documentation, generally _work_, etc. 

Things were going _mostly_ swimmingly until I went to implement a particular item on my checklist:  MAVLink 2 extended messages. The problem is as follows: the XML format currently used to identify extended message parameters does not annotate the extended messages in any meaningful way _unless you are parsing them sequentially._ Most XML parsers allow you to use something like XPath (or in Go's case, define `structs` with XPath-ish annotations, and parse data into them reflectively) to select nodes by their attributes, children, ancestors, etc. By injecting the `<extensions/>` node as a sibling of the `<field>` nodes, the parser must maintain state in order to identify a `<field>` as being an extension. While it's not a show-stopper for me (I got around it by rewriting it with a regex), it's probably going to limit others that wish to follow in my footsteps, and it's not semantically-valid XML.

Currently, the XML definition looks like this (pulled from [here](https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md)):

```xml
  <message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
    <extensions/>
    <field name="extrafield1" type="uint8_t">extension1</field>
    <field name="extrafield2" type="uint32_t">extension2</field>
  </message>
```

I would suggest moving to a format that annotates the fields either directly or hierarchically. A hierarchical annotation might look like the following:

```xml
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
    <extensions>
        <field name="extrafield1" type="uint8_t">extension1</field>
        <field name="extrafield2" type="uint32_t">extension2</field>
    </extensions>
</message>
```

while a direct annotation could simply be the addition of an attribute to the field, like so:

```
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
    <field name="extrafield1" type="uint8_t" extension="true">extension1</field>
    <field name="extrafield2" type="uint32_t" extension="true">extension2</field>
</message>
```

If you wanted to go one step further, it might be nice to add an attribute to the `<extensions>` node indicating the minimum version for which the extensions are applicable (i.e. `<extensions version="2">`) and then using multiple `<extensions>` nodes to group child `<field>`s by minimum version – or use the `extension` attribute of the `<field>` node to contain that information.

Just my two cents – hoping to save someone else this headache! Thanks for all your hard work. This is a great project 👍

kenkeiter, 2017-11-14:
As a side note, if anyone's wondering how I'm dealing with this using a regex: https://play.golang.org/p/TbkDWmadjd

amilcarlucas, 2018-02-19:
These are very good points. What do you think @LorenzMeier  ?

hamishwillee, 2018-02-19:
@mrpollo FYI.

LorenzMeier, 2018-05-26:
Closing as stale.

amilcarlucas, 2018-05-28:
Why closing this ? The idea is very good, for example:

'''
<extensions version="2">
...
</extensions>
<extensions version="3">
...
</extensions>
'''

WickedShell, 2018-05-28:
Yeah, someone would need to update the pymavlink stuff for the change, (and I will need to update the [Clojure parser](https://github.com/wickedshell/clj-mavlink)) but the current `<extension>` tag structure imposes a lot of unneeded awkwardness when making a generator.

LorenzMeier, 2018-05-28:
Background for closing this issue: I'm trying to enable a culture where we focus on the issues with engagement or actual contributions. I realize that I should have provided the rationale in the first place, but it is here now:
https://github.com/mavlink/mavlink/issues/925

Could you please send a PR instead so we have code to look at? Its ok if its not functional, but that makes it easier to discuss and we're generally not closing stale PRs at the same rate as stale issues but try to resurrect them.

WickedShell, 2018-05-28:
@LorenzMeier he did provide an inline example of what it looks like. I think everything needed to evaluate it (other then the pymavlink changes) can be considered present.

hamishwillee, 2018-05-28:
@WickedShell I opened another issue some days ago to discuss versioning - #919. This issue is lively now, so there is no reason we can't discuss it here, but we shouldn't have both ;-0. I agree with you in https://github.com/mavlink/mavlink/issues/795#issuecomment-392448420 - we're not ready for a PR. 

I think the below works fine for "introduced in", but what about for removal? Say I add extension in mavlink 2, deprecate in 3, remove in MAVLink 4. How do I represent that?
```
<extensions version="2">

...
</extensions>
<extensions version="3">
...
</extensions>
```
My gut feeling is that this would work better on a per-field basis 
```
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
    <field name="extrafield1" type="uint8_t" extension_start="mav2">extension1</field>
    <field name="extrafield2" type="uint32_t" extension_start="mav2" extension_end="mav4">extension2</field>
</message>
```
Or maybe a combination
```
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
   <extension version="2">
       <field name="extrafield1" type="uint8_t" removed="mav3">extension1</field>
       <field name="extrafield2" type="uint32_t">extension2</field>
     </extension>
</message>
```

kenkeiter, 2018-05-28:
Glad to see that this issue has been reopened! 

@hamishwillee TBQH, I would go for the first version, in which extensions are enumerated separately for each version (no version-from/version-to). Definition file size doesn't seem to be as much of a concern as readability and ease of codegen, in this application.

hamishwillee, 2018-05-28:
@kenkeiter I take your point, but with that approach there is no way to ever remove a message - because if you do so you'd be removing even from versions in which it is supported.

I'm leaning towards the combination approach above - essentially grouping to indicate when a feature is introduced, with an optional per-field flag to indicate the version that an extension has been removed. 
```
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
   <extension version="2">
       <field name="extrafield1" type="uint8_t" removed="mav3">extension1</field>
       <field name="extrafield2" type="uint32_t">extension2</field>
     </extension>
</message>
```
Presumably we'd have to add versioning to deprecation as well.

kenkeiter, 2018-05-28:
I think we're suggesting almost the same thing – except that I'm suggesting a full enumeration of all extensions per version. So, for example:

```
<message id="152" name="MEMINFO">
    <description>state of APM memory</description>
    <field name="brkval" type="uint16_t">heap top</field>
    <field name="freemem" type="uint16_t">free memory</field>
    <extension version="2">
        <field name="extrafield1" type="uint8_t">extension1</field>
        <field name="extrafield2" type="uint32_t">extension2</field>
    </extension>
    <extension version="3">
        <!-- In version 3, extrafield1 is omitted – it's been removed -->
        <field name="extrafield2" type="uint32_t">extension2</field>
    </extension>
</message>
```

It is more verbose, but:

1.  _certainly_ easier to write codegen for;
2. the parser contains less state, since it doesn't need to parse all extension fields to determine which fields exist at a specific version;
3. it's easier for humans to read/parse visually;
4. the verbosity shouldn't matter, since the files are edited relatively infrequently.

hamishwillee, 2018-05-29:
@kenkeiter Thanks for that clarification. Yes that sorts out the problem and plays nicely with deprecated and other proposed tags. 

I can see that the parser would be much easier to write, but I do not like the verbosity. 
- Every time a version is updated, there is a copy paste of previous section across a whole doc rather than a few minor tweaks.
- If anything changes - e.g. someone decides to add full stops to description - it needs to be added to every iteration of the field. 
- It is hard for users to see the difference between versions by inspection. Ie I don't think it is actually easier for humans to parse
- I think we're far more likely to add fields than remove them. To me that verbosity is then a problem.

But I don't have to write the parser :-)

@amilcarlucas @WickedShell @LorenzMeier Do you have a strong opinion? The main choices are outlined above in the last two comments:
- https://github.com/mavlink/mavlink/issues/795#issuecomment-392618837
- https://github.com/mavlink/mavlink/issues/795#issuecomment-392617312
- Any other options?

amilcarlucas, 2018-05-31:
I do not think that we should remove fields. Period.

So... I agree with it but I think any new version should only add new fields, this avoids code duplication.
If we need to change the message, deprecate it and create a new one.

But the main point is .... make sure to think twice before changing the standard. This prevents having to modify it later :)

kenkeiter, 2018-05-31:
@amilcarlucas That's a really good point – maybe it's worth taking an approach more like Google's with protocol buffers?

hamishwillee, 2018-05-31:
@amilcarlucas  Hmmm. Not removing things offends my sense of tidiness, but may make sense. I think we probably are near point where we should discuss in mavlink meeting and agree way forward. 

On the other hand, if deletion isn't required we only probably need new extensions in each group (rather than all of them). Deprecation fits in as a version tag?

amilcarlucas, 2018-06-01:
@kenkeiter google's Protocol buffers, would be a complete replacement for mavlink, but it is missing some mavlink features, so that is not an option. And as far as I can see makes no sense combining both.

So ... we need to extend mavlink.

@hamishwillee Removing things is still possible, by deprecating them! But you deprecate a message, not a field inside a message. It is not meant to be fine granular. The line has to be drawn at some point.

I think deprecation should be attached to a date, not to a version.

hamishwillee, 2018-07-04:
FYI, This has stalled. I have added versioning as a possible topic for discussion in the meeting next week [20180711-Weekly-Meeting](https://github.com/mavlink/mavlink/wiki/20180711-Weekly-Meeting).

The whole versioning and identifier thing is quite unclear:
- MAVLink Protocol  version 1 and 2
- Message versions
  - We have a folder 0.9 or 1.0
  - We have a version number in each xml file.
    ```
    <mavlink>
    <version>3</version>
    <dialect>0</dialect>
    ```
    - Is the version supposed to be the third iteration of this file - ie should it be updated with every new message or change to any message? If not, when/why does this get updated.
  - We have a dialect in each file. How do dialects map to versions - ie if we specify a dialect that depends on another file, can we force the version that it depends on? Or does the dialect change.
  - We have `<extensions>` which basically says that these parts of these fields apply only after MAVLink 

From all the above I *think*:
1. Everything uses the 1.0 folder for messages
1. The only real versioning we currently have is use of `<extensions>` to indicate that something is for MAVLink2 only. 
1. The dialect number is the unique identifier for a set of messages. You can version a dialect, but that only applies to its own messages - any included messages will be the latest in the included file
1. I think maybe everything works for "compatibility" because we only make additions - and newer versions simply ignore messages they can't handle.  

Perhaps taking us backwards, but I was thinking we possibly need to start with a set of problems we are trying to address/requirements that we'd like to meet on versioning so we can validate any solution suggestion. Normally the main requirement for versioning is compatibility. 

Would be great if someone could help me understand better what we have and how it is used by the generated library.

benbarkay, 2018-07-05:
In terms dependency structure, my view make/model dialects shouldn't inherit, they should be composed.
Imagine the following layout:
1. There are different packages of messages: `mission.xml`, `mission-beizer.xml`, `mission-...` etc, where in this example, `mission-beizer.xml` would depend on `mission.xml`.
2. There are packages for make/model, e.g. `ardupilotmega.xml`:
```xml
<mavlink>
    ....
    <messages>
        <include src="mission/mission.xml" />
        <include src="mission/mission-beizer.xml" />
        <!-- ... -->
    </messages>
</mavlink>
```
I'd even go forward to suggest that perhaps we could use GAVC for these packages, and use a Maven repository:
```xml
<mavlink>
    <messages>
        <include gavc="org.mavlink.messages:mission:1.1.2" />
        <include gavc="org.mavlink.messages:mission-beizer:1.1.5" />
        <include gavc="io.dronefleet.mavlink:led-draw:1.0.1" />
        <!-- .. -->
    </messages>
</mavlink>
```
This would allow users to do whatever they want without having to push it to this repository. It would also allow different dialects to use *portions* of what's already out there, instead of having it as part of their stack and then apologizing that it isn't implemented.

Obviously the above is just an illustration. But the main point I am trying to get to here is that the design error in what there currently is, is the same as composition over inheritance (where here we only allow inheritance).

Vavooon, 2018-07-05:
That's great that you've started work on versioning.

First of all I want to say that processing messages from various Mavlink devices is not always a trivial task since there is no info what exact Mavlink version is used. For example, I faced an issue when at some point _target_system_ and _target_component_ fields were added to **COMMAND_ACK** message. 
My app which was compiled with latest Mavlink library tried to send the message (received from latest ArduCopter which still uses 1.5 year old Mavlink) to exact system and component instead of broadcasting it, thus message was not delivered to GCS.

I saw that Lorenz Meier have added [PROTOCOL_VERSION](https://github.com/mavlink/mavlink/commit/e7221f41d3d43b432c26a0369d867d86f559ee42) message which may be useful for determining which exact Mavlink version is used by a device.
There is a _version_ field which stores protocol version * 100. So if we decide to increase version on 1 each time we change some message we will have 100 commits until we reach next major version. Another idea is to bump it each month or so or when major changes were added.

I'd like to see _removed_ field since it may be useful when we want to change field data type:
```
<field name="extrafield1" type="uint8_t" version="273" removed="300">extension1</field>
<field name="extrafield1" type="uint32_t" version="300">extension1</field>
```

However I realize that such possibility will make generated libraries more complex, requiring something like _QVariant_:
```
uint32_t value = systemStatusMsg.voltage.toUint32();
```

hamishwillee, 2018-07-05:
@benbarkay I like the idea of composition, as a partial solution to the problem of working out what messages a mavlink system actually supports. Not sure about the actual infrastructure, but would want it to know about versions of the composed classes. I guess this adds complexity in managing the ID space.

@Vavooon Yes, we need better versioning.  I've added this for discussion in the next dev meeting for mavlink.

benbarkay, 2018-07-06:
@hamishwillee that will always be a burden until we find a good automated way to do it. I propose though that it's up to the leaf dialect (make/model) to define the ID ranges for messages. We could throw a compile-time error (or runtime from the perspective of the generator) if ID ranges clash. Either that or completely automate the ID generation when a flat XML is generated for a dialect.

If you'd like the message IDs to be unique, you have three options:
1. Either associate them with a namespace which is guaranteed to be unique, e.g. GAVC for Maven Central (you can't have two of those which clash) along with an internal ID for each message in the package, or
2. Have everyone manually synchronize with eachother (this is what is currently happening), or
3. Define a private authority policy, such as the owner of the leaf dialect, allowing them to inject message IDs to the messages that they are including. This is the user of the message packages.

hamishwillee, 2018-07-09:
@benbarkay 

I don't really want to say anything much here. I'm frightened of making myself look like an idiot though ignorance. Doing lots more reading/research on MAVLink now, so may have something useful to contribute in the coming weeks. 

IN the meantime, at the risk of saying something stupid/obvious:
- Whatever we do, we will need to have a core set of invariant message ids to reduce the effort for GCS/higher level API developers, irrespective of how we compose the messages we build. 
- Maybe we could allow each dialect to define its own range in XML , but the "actual" generated ID would be the `dialect number + message id`.

benbarkay, 2018-07-09:
We're all born idiots, those who ridicule the mistakes of others in my mind are the ridiculous ones :)

In terms of the issue at hand, using dialect numbers still pose a problem, because they could collide, and that means that there has to be a central authority that will hand them out. Currently, it's this repository, and in the future we could perhaps have a dialect registry -- which, if automated, might mean that people will abuse that in some ways (reserving IDs, requiring IDs for unused dialects etc), which will then put us in a need for policy which would be hard to form.

I propose that we look for a solution that would decentralise dialect/message id spaces, to where they cannot conflict by design. That could happen if we decouple dialects and isolate them as packages, allowing the outer-scope, dependent packages, to define the ID for each message. So a Heartbeat in my dialect could have the ID of 10, but in your dialect it's 0, and it's totally fine -- because our dialects decide on the IDs of the messages. Once a dialect is resolved, then you also know the IDs of the different messages, it also doesn't have to be manually selected, it could be part of the generation process.

Hopefully I've made my idea clearer, or made it clearer which part I don't understand in what you are trying to tell me :)

LorenzMeier, 2018-07-09:
The core value of MAVLink is the standardization, not the technology itself. When people in the industry talk about MAVLink they see the same value as they see in USB. So I'm rather sceptical about investing a ton of effort into supporting more fragmentation.

amilcarlucas, 2018-07-09:
I also do not see added value in the fragmentation.
- Just add a process to deprecate entire messages.
- and a process to make extensions with a <extension = x>.....</extension>
And that is it.
But with a 24bit address space I do like the idea of assigning large ID sections to a dialect.
Something like 2048 IDs per dialect. That should avoid collisions in the next couple of years.

But yes the core of the work needs to be done in common.xml. That way all benefit form the improvements.

benbarkay, 2018-07-09:
I'll explain the vast difference between the effects of what you're doing, and USB then.

USB is a strict standard which allows *me* to *implement*, for all intents and purposes, whatever I like on top of its abstraction. I can implement Mavlink over USB, without needing to create a pull request to any centralised USB repository, or worry about any message collisions.

On the other hand, what you guys are doing is you are closing Mavlink to implementors. Mavlink (currently) is not an abstraction layer, such as USB is. I can use USB for my implementations without modifying USB, in the same way that I use TCP or HTTP or any other widely successful protocol.

It is true that I can diverge Mavlink, and provide my own implementations of it, but I cannot live alongside other implementations that use Mavlink. For example, if I wanted to create a Mavlink router, that would have been a good use-case for loading dialect configuration files (in the same fashion that you would install a driver for a USB component) and having the router route to a destination, validating packets and such. This cannot be achieved without coordinating all of the projects that use Mavlink, because of Mavlink's current design.

"Fragmentation" (I don't understand what that even means here) of this kind, which I think should actually be called decoupling, is the reason that you can do things like offer your Mavlink library without having to ask Microsoft, Apple, The Linux Foundation, the FreeBSD project and the many more to allocate spaces (ID, name, memory or whatever) for you so that users could enjoy your labor. It decentralises those things.

I may still not understand what it is that you mean, or perhaps I am not there yet (or maybe never will be) in terms of your understanding of how this market/engineering principles/concept/idea works like you do, but for what it's worth I tried my best to push forward my view. All in good spirit :)

I shall now cease discussion space consumption in this issue.

LorenzMeier, 2018-07-09:
Hi Ben - that's why I said the people in industry - what you're describing is a customisation / prototyping value that I'm not disputing. The main value of USB is in the common classes:
http://www.usb.org/developers/docs/devclass_docs/

This is what the USB forum agrees on and there is a single standard for e.g. webcams that want to be plug&play capable. This is why a USB memory stick is working in every computer. There is only one way to do it - like common.xml.

You still can bring your own driver and do your own device, but that way you loose plug&play capability.

With a proprietary USB device you need to have custom code on the device side and custom side on the PC side and all what USB is offering is transport. You need even to load a kernel module to enable that device. You need to install and distribute that kernel module. It gets less testing than a common driver. That is arguably similar (however not exactly the same) for a MAVLink dialect. MAVLink dialects do not scale as well as USB drivers do, but then MAVLink is used in a much more embedded domain. We have been discussing allowing dialects to co-exist on developer meetings and calls, which would make MAVLink an almost 1:1 mapping to how USB is filling needs of an interoperable ecosystem.

MAVLink can transport unknown messages (it requires the right configuration) like USB does and act like a pure transport and hand off decoding of messages to a higher layer. But then you would need to first somehow install the "driver" for that custom logic.

benbarkay, 2018-07-09:
@LorenzMeier I agree with most of your points, but I think we're missing a dimension here.

What I am pointing at is that a driver is not part of USB as a component. I do not need to update my USB host or peripheral firmware in order to use a driver. A driver to me, metaphorically to Mavlink, would be an XML definition file. We could have a common XML file, or hopefully, multiple different common XML definition files for different parts/topics, and allow users of Mavlink to choose to make their implementations for these and benefit from great integration with other tools. The problem is not that, it is in that we do not allow a workable way to let third-party so-caller "drivers" to integrate with Mavlink. This difference to USB is the one thing that I am in argument with. I have no problem with the benefit of allowing standard "drivers" or standard xml files that many implementors can support, what I do have a problem with, however, is that we are closed in terms of allowing third-party ones to live alongside these standard ones, requiring them to perform a "firmware upgrade" to whatever they want to talk to instead of simply providing their XML file.

If I want to use Mavlink to implement communication for my smart home these days, I will find myself doing so with a repository that is forked from yours, and whatever middleware infrastructure there is, such as a hypothetic Mavlink router will not be able to work with it unless it receives a firmware upgrade from a build from my repository.

benbarkay, 2018-07-09:
@LorenzMeier Actually, I've changed my mind in retracting... Seeing as you use this USB argument a lot, it may be a good idea to put this forward... Would like to clear though that this is all in good spirit, I am thankful for Mavlink as it is despite my wishes for improvement :)

I agree with most if not all of your points above, but I think they're evading the dimension that I am attempting to address here.

The problem is not with the standardization part, it is in that we do not allow a workable way to let third-party so-called "drivers" to integrate with Mavlink, privately.

Let me list concrete usability differences between Mavlink and USB, and perhaps these multiple vectors will yield the pain point:
1. I can place a USB hub (or any USB middleware for that matter) between a device and a host machine. I do not need to update the firmware of my USB hub every time that I want to use a different implementation that uses USB. That is not the case with Mavlink. A Mavlink hub or router would have to understand the packets, and have to keep receiving "driver" updates.
2. I can publish an implementation that uses USB without making a pull request to the official USB specification without worry. This is not the case with Mavlink. If I do not make a pull request to extend the official specification, at some point, someone might (or, eventually will, if Mavlink becomes successful) consume the message ID space (or any other ID space for that matter) that I chose for my product.

Obviously, the implementations that do choose to use plug-and-play features of USB will need to integrate with what's in the official USB specification. But also staying *open* for extension without having to upgrade firmware yields capabilities such as the ability to add features for systems with drivers and falling back to common spec when no driver is present, or making middleware that could support these driver-based devices.

If you look at the progression of USB, you will find out that in the beginning, USB didn't have much to offer except mass storage and limited mouse/keyboard support if you exclude the third-party drivers. Printers had to have drivers installed, cameras had to have drivers installed, wifi adapters had to have drivers installed etc... It is only after there were ample third party implementations that it was visible how to intersect a bunch of these into a unified spec. These things need to be developed simultaneously, as they have the aforementioned symbiotic relationship.

Now just like any analogy, there are analogous parts and parts that aren't analogous. Discussing a determination of which parts in the comparison with USB matter for Mavlink and which parts don't would be too involved and mazey, but I think that if USB had gone out without allowing privately published third-party drivers, it would have been so far away from what it is today, to a point that perhaps it wouldn't be as widely spread for you to provide it as a standard for a successful platform.

amilcarlucas, 2019-10-26:
So .... now we do have `deprecated` and `wip` tags in place, so that part is covered.

How about discussing replacing : 
`<extensions/>` 
with:
`<extensions version=2>` 
...
`</extensions>`

hamishwillee, 2019-10-27:
Re https://github.com/mavlink/mavlink/issues/795#issuecomment-546630250

@amilcarlucas I'm not sure there is any need. Provided all extensions are additive the messages remain backward compatible. The reason you might want this is to be able to test whether the new version is supported, but you don't need to know that as a user if the extension is properly designed, because you set the default of 0 to mean "not a valid value" and if you get a valid value you can work with it if you don't get zero.

The bigger picture of managing versioning is probably better fixed using [Microservices versioning](https://docs.google.com/document/d/1zK_JcJbhtbRA3W0MgnXSNWMUxOcBu6vXocrjZVMXki4/edit?pli=1#heading=h.t3uczm564nmc)

hamishwillee, 2020-06-22:
Closing for reasons provided previously

closed 2020-06-22 by hamishwillee