mavlink/mavlink #797: importing x25crc in Python implementation

jxltom, 2017-11-14:
Currently, the python mavlink implementation generated by mavgenerator.py contains ```from ...generator.mavcrc import x25crc```.

I have to including mavcrc module manually to make the generated script running, is it better to including the ```x25crc``` module into the generated python implementation directly?

amilcarlucas, 2018-02-20:
Please open a pull request rather than an issue. It is easier to discuss code modifications using a PR

amilcarlucas, 2018-02-20:
This can IMHO be closed now

hamishwillee, 2018-02-20:
@jxltom Do you think you could make this PR? I agree with @amilcarlucas that it would be best if you did this, but if not we should probably keep the issue open.

jxltom, 2018-02-22:
I found that this is a upstream issue. So I made a PR in https://github.com/ArduPilot/pymavlink/pull/152. We can discuss the code there.

After that PR is merged, sync upstream and update submodule in this project will solve this problem.

closed 2018-02-22 by jxltom