mavlink/mavlink #802: Cannot use MavUtils with custom dialect

JesperChristensen89, 2017-11-21:
I am able to generate the needed files for MAVLink in C with only a few custom messages (no common etc.).
However, for Python I can't get the same routine to work. I can generate the .py file with my messages, but I still want to utilize some functions from mavutils, e.g. serial connection. How do I achieve this? I do not need any ardupilot or other dialect, just 1 I will define myself with a few custom messages.

amilcarlucas, 2018-02-19:
You need to see how ardupilotmega.xml does it and do something similar for your own defenition file.
And do not include common.h directly, but the file header generated for your dialect instead.

This can be closed now AFAIK

hamishwillee, 2018-02-20:
Closing. @JesperChristensen89 if you need to discuss further perhaps raise on Slack: https://mavlink.io/en/#support

closed 2018-02-20 by hamishwillee