mavlink/mavlink #803: SERVO_OUTPUT_RAW isn't working properly

patrickelectric, 2017-11-21:
I have done some tests with ArduPlane and ArduSub, it appears that SERVO_OUTPUT_RAW message have problems to create the mavlink header.

The only workaround to this problem is to force the channel to use mavlink2.
But..:
`SerialProtocol_MAVLink2 = 2, // do not use - use MAVLink and provide instance of 1`
Any idea why this change is necessary ?
I have all SERIALX_PROTOCOLS in MAVLink2.

Checking the problem:
```cpp
mavlink_status_t *status = mavlink_get_channel_status(chan);
// If mavlink1 is True, it'll be used mavlinkV1 header.
bool mavlink1 = (status->flags & MAVLINK_STATUS_FLAG_OUT_MAVLINK1) != 0;
```
Forcing mavlink1 to be false and using mavlink2 work without problem.

----
Picture of the values, graph and variance.
[image]

I'm using ardupilot master with mavlink a8bb1bf5d4915862eb9c6f4c8ad6cdb7acee56ef

amilcarlucas, 2018-03-08:
Are you using `--mav20` in mavproxy

LorenzMeier, 2018-05-26:
Please file an issue in the Firmware repo.

closed 2018-05-26 by LorenzMeier