mavlink/mavlink #805: Java generator hard-codes sysid and compid

jacobsenmd, 2017-11-23:
The Java code generator hard-codes a sysid of 255 and compid of 190, overwriting a user-specified sysid or compid. For example, this is the msg_heartbeat packing function:

```
public MAVLinkPacket pack(){
        MAVLinkPacket packet = new MAVLinkPacket(MAVLINK_MSG_LENGTH);
        packet.sysid = 255;
        packet.compid = 190;
        packet.msgid = MAVLINK_MSG_ID_HEARTBEAT;
              
        packet.payload.putUnsignedInt(custom_mode);
              
        packet.payload.putUnsignedByte(type);
              
        packet.payload.putUnsignedByte(autopilot);
              
        packet.payload.putUnsignedByte(base_mode);
              
        packet.payload.putUnsignedByte(system_status);
              
        packet.payload.putUnsignedByte(mavlink_version);
        
        return packet;
    }
```

This was most likely implemented because Java is typically used for a GCS, but that is not always the case. In my case, I am creating a mock drone in Java. It makes sense that these values could be used as defaults, but should not be hard-coded into packing functions.

My recommendation is to make these defaults in MAVLinkMessage, and remove them from the subclassed message packing functions.

amilcarlucas, 2018-02-20:
@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here.

hamishwillee, 2018-02-20:
@jacobsenmd Can you please report upstream as per https://github.com/mavlink/mavlink/issues/805#issuecomment-366965361

jacobsenmd, 2018-03-14:
Moved to https://github.com/ardupilot/pymavlink/issues/160

closed 2018-03-14 by jacobsenmd