mavlink/mavlink #806: Java mission messages throw exception

jacobsenmd, 2017-11-23:
Several of the mission messages generated in Java throw IndexOutOfBoundsException when unpacked. So far I have seen this behavior with msg_mission_count, msg_mission_item, and msg_mission_ack.

For example, msg_mission_count.unpack():
```
public void unpack(MAVLinkPayload payload) {
        payload.resetIndex();
              
        this.count = payload.getUnsignedShort();
              
        this.target_system = payload.getUnsignedByte();
              
        this.target_component = payload.getUnsignedByte();
              
        this.mission_type = payload.getUnsignedByte();
    }
```

The final line throws the following exception (as recorded in Android adb):
```
11-22 16:32:23.321 10755-10882/io.diux.rogue.rosettadrone D/Rosetta: exception
                                                                     java.lang.IndexOutOfBoundsException: index=4 out of bounds (limit=4)
                                                                         at java.nio.Buffer.checkIndex(Buffer.java:528)
                                                                         at java.nio.HeapByteBuffer.get(HeapByteBuffer.java:118)
                                                                         at com.MAVLink.Messages.MAVLinkPayload.getUnsignedByte(MAVLinkPayload.java:62)
                                                                         at com.MAVLink.common.msg_mission_count.unpack(msg_mission_count.java:81)
                                                                         at com.MAVLink.common.msg_mission_count.<init>(msg_mission_count.java:100)
                                                                         at com.MAVLink.MAVLinkPacket.unpack(MAVLinkPacket.java:420)
```

amilcarlucas, 2018-02-20:
@hamishwillee This is a pymavlink issue that should be reported upstream to https://github.com/ardupilot/pymavlink/issues and closed here.

hamishwillee, 2018-02-20:
@jacobsenmd Can you please report upstream as per https://github.com/mavlink/mavlink/issues/806#issuecomment-366965216 ?

jacobsenmd, 2018-03-14:
Moved to https://github.com/ArduPilot/pymavlink/issues/new

WickedShell, 2018-03-14:
This isn't actually an upstream issue. This looks to be associated with the incomplete MAVLink2 changes which were brought in on this fork.

LorenzMeier, 2018-03-14:
@WickedShell It would be very kind if you could you send a PR to address that and keep things in sync.

WickedShell, 2018-03-14:
@LorenzMeier It's not my job to correct a bad patch set that was brought in on a downstream fork which I'm not associated with, and was merged without a thorough review.

Please reopen the issue and remove the upstream label as it is still a relevant issue here.

WickedShell, 2018-03-14:
The problem does not exist in upstream. Upstream doesn't support MAVLink2 yet, as the patches either need to be tested, or the PR's for it had flaws the authors would not address.

jacobsenmd, 2018-03-14:
I have spoken several times with both communities about the immense frustration caused by forking Mavlink's implementation. This will only continue to get worse. Please consider this another data point, and a plea to the leadership of both sides to find a long-term solution. It doesn't do any good to pretend that the communities are independent, when code is entangled at such a low level. I work with a number of different projects that need to support various flavors of mavlink, and having two divergent implementations with the same names and package identifiers is wreaking havoc.

hamishwillee, 2018-03-14:
Reopening for reasons given in https://github.com/mavlink/mavlink/issues/806#issuecomment-373151842

hamishwillee, 2020-04-23:
The fork no longer exists. Closing.

closed 2020-04-23 by hamishwillee