mavlink/mavlink #811: Invalid MAVLink messages from example mavlink_udp.c?

hdino, 2017-12-20:
While running the example I noticed that APM Planner does not process the generated MAVLink messages. Hence, I printed the serialised heartbeat message using MAVLink v2 headers:
```
mavlink_message_t msg;
mavlink_msg_heartbeat_pack(1, 200, &msg, MAV_TYPE_HELICOPTER, MAV_AUTOPILOT_GENERIC, MAV_MODE_GUIDED_ARMED, 0, MAV_STATE_ACTIVE);
len = mavlink_msg_to_send_buffer(buf, &msg);
for (size_t i = 0; i < len; ++i)
   printf("%02X ", buf[i]);
```
Result: `FD 09 00 00 00 01 C8 00 00 00 00 00 00 00 04 00 D8 04 03 E4 9B`

I was a bit confused since the magic number (0x55) does not appear in this line. Therefore, I ran the SITL which sends the following line:
`fe 0c bc 01 01 02 60 f6 b6 19 c8 60 05 00 06 41 01 00 95 0d`
There is also no 0x55, but it seems to be a valid MAVLink message.

This raises two questions:
- Why does the short code example generate invalid messages?
- Why is there no 0x55?

magicrub, 2017-12-20:
What magic munmber 0x55 are you talking about?
MAVLink v1 start packet byte: 0xFE (I see this in your result)
MAVLink v2 start packet byte: 0xFD (I see this in your SITL packet. v2 is default until someone talks v1 to it then it reverts to v1)

hdino, 2017-12-21:
Oh, I see 0x55 was used prior to MAVLink v1... Thank you for pointing this out. In addition, I figured out that APM Planner apparently does not support MAVLink v2 and thus did not respond to my v2 messages...

closed 2017-12-21 by hdino