mavlink/mavlink #815: Header problem with compiler

tuloski, 2017-12-27:
Using the generated .c files for the v2.0 I get an error when compiling at common.h.
The error I get is: "error: braces around scalar initializer for type 'const uint8_t {aka const unsigned char}' for the line `#define MAVLINK_MESSAGE_CRCS {{0, 50, 9, 0, 0, 0}, {1, 124, 31, 0, 0, 0}, {2, 137, 12, 0.....`.

Probably that initialization is not supported by all compilers (I wasn't able to find many info), so a more supported initialization would be better. 

No problems at all with v1.0, where the CRC was a single array, instead of a two dimensional array.

vooon, 2017-12-27:
It is not 2-dim array, it is array of `mavlink_msg_entry_t`.

tuloski, 2017-12-27:
Ok, but knowing that doesn't solve my problem.
I'm compiling it on arduino FYI.

magicrub, 2017-12-27:
@tuloski  Can you try this generator version and see if it works?
https://github.com/ArduPilot/pymavlink

tuloski, 2017-12-27:
What's the difference from https://github.com/mavlink/mavlink/? I used the latter, but I think it's the same. Would that generate the MAVLINK_MESSAGE_CRCS without that strange array of mavlink_msg_entry_t?

magicrub, 2017-12-27:
mavlink/mavlink hsa a submodule of pymavlink. Notice at the top it says mavlink/pymavlink forked from ArduPilot/pymavlink.
https://github.com/mavlink/pymavlink/tree/7f090dcbec140a3e07ab9a2454beba1d948a7b86

The mavlink one is older, it just needs an update from the ArduPilot fork. Easy for the mavlink maintainers to do, they just need to do it to stay up-to-date. It's not far behind, just a few months. This is common for submodules

tuloski, 2017-12-28:
Sorry but there is no usage information for the generator in the repo you mention.
Also I see that the CS generator supports again only v1.0 (different issue, but it is a symptom of non-updated code).

tuloski, 2017-12-28:
EDIT: I was able to use pymavlink, but the MAVLINK_MESSAGE_CRCS has the same structure, so I get the same error.

magicrub, 2017-12-28:
Is your Arduino compiler using gcc? What version?

amilcarlucas, 2018-02-19:
I have updated the forked mavlink/pymavlink repository.
It proabbly helped with this problem.
@tuloski can this be closed now ?