mavlink/mavlink #816: Confusion with protocols.

tuloski, 2017-12-28:
I see lot of confusion with the different protocols, but maybe it's only me.

- Here http://mavlink.org/messages/common it says that the protocol version is 3.
- Usually all the programs support v1.0 and v2.0
- The set of messages .xlm here https://github.com/mavlink/mavlink/  is referred as 0.9 and 1.0

What are all those numbers and confusion, why 0,9, 1.0, 2.0 3...?

LorenzMeier, 2017-12-28:
Current MAVLink version is v2.0, the protocol version you're referring to is a sub revision (so effectively 2.3). We'll fix that in the documentation, thanks!

closed 2017-12-28 by LorenzMeier

tuloski, 2017-12-28:
In the documentation here http://qgroundcontrol.org/mavlink/start also the packet anatomy is of the old v1.0, but there is not for v2.0.

hamishwillee, 2017-12-28:
Thanks @LorenzMeier. FYI this also has now propagated through to the devguide (https://github.com/mavlink/mavlink-devguide/pull/30) - we still have some work to do in Jenkins to automate rebuild for this sort of change.

@tuloski The links you point to are deprecated. The "official" docs are now hosted at https://mavlink.io/en/ - for example the [common.xml is here](https://mavlink.io/en/messages/common.html).

For the package anatomy see that we have both version packet formats/documented here: https://mavlink.io/en/protocol/overview.html

I am supposed to be migrating the remainder of the old MAVLink docs to the new site and removing the old versions. However there is lots to do, and this is not my highest priority.