mavlink/mavlink #828: How should camera trigger be handled in simulation?

kjetilkjeka, 2018-01-21:
A common application for UAV's is surveying missions, it's clear how MavLink accommodates to these use cases. Another MavLink use case is sending "mocked" sensor measurements and states between the simulator and SIL/HIL. How to work with these use cases simultaneously is not entirely clear to me.

What i specifically want to achieve is planning a mission with QGC (that includes camera triggering), uploading it into PX4 and having images captured from AirSim as waypoint are reached.

The problem I'm having is that i don't know what packet i should send to the simulator from PX4, the alternatives i see is.

##### 1. `COMMAND_LONG` with `MAV_CMD_IMAGE_START_CAPTURE` (`MAV_CMD_DO_DIGICAM_CONTROL`)

It seems to break abstraction as this message is not intended for use in HIL. However, PX4 SITL already uses this with `MAV_CMD_SET_MESSAGE_INTERVAL` so it might be the best choice.

##### 2. `CAMERA_TRIGGER`

Seems to be a message intended for timestamping rather than actually ordering a trigger to occur.

##### 3. A new `HIL_CAMERA_TRIGGER`, `HIL_COMMAND_LONG` (or something similar) message.

Create a new message type similar to the other HIL message types made specially for use in SIL/HIL.

hamishwillee, 2018-01-29:
Something similar was just done to integrate camera into Gazebo - @ndepal are you the right person to comment on how this should be handled?

hamishwillee, 2018-01-29:
@dogmaphobic This raises a question for me about the [camera protocol](https://mavlink.io/en/protocol/camera.html) docs.  This doesn't cover MAV_CMD_DO_SET_CAM_TRIGG_DIST and CAMERA_TRIGGER at all 
- I believe those are the "old" API?
- Does PX4 understand the new protocol, and in what version?
- Does it still understand those old messages?
- If a GCS requests an image using the new protocol, how does it "get to" the simulated camera? Is the simulator also on the network somehow? Does PX4 actually have to do anything to forward messages to "component MAVLink cameras"?

ndepal, 2018-01-29:
My involvement with what @hamishwillee is referring to was only on the PX4 heading control side, so I can't comment on what message should be used for PX4->simulator communication for this.

kjetilkjeka, 2018-01-29:
> If a GCS requests an image using the new protocol, how does it "get to" the simulated camera? Is the simulator also on the network somehow? Does PX4 actually have to do anything to forward messages to "component MAVLink cameras"?

For PX4, it doesn't reach the simulated cameras at all. This is what i want to implement. Since the MavLink interface responsible for SIL/HIL is seperated from the other MavLink interfaces [as described here](https://dev.px4.io/en/simulation/#sitl-simulation-environment) the [px4 simulator module](https://github.com/PX4/Firmware/tree/master/src/modules/simulator) would have to forward it to the simulator.

hamishwillee, 2018-01-30:
@kjetilkjeka I propose this can be closed as it has been answered in https://github.com/PX4/Firmware/pull/8775#issuecomment-361476205 . OK?

@ndepal Thanks!

kjetilkjeka, 2018-01-30:
I agree @hamishwillee

closed 2018-01-30 by kjetilkjeka