mavlink/mavlink #830: Common protocol: no mission stop/abort

tuloski, 2018-01-26:
It seems to me that there isn't a mavlink common message or a MAV_CMD to stop or abort a mission and I think it is foundamental at protocol level.
What's the most used protocol in case of mission abort now in the most common platforms (arducopter, pixhawk, mission planner...)?

magicrub, 2018-01-26:
What does "stop" mean and what behavior do you expect?
What does "abort" mean and what behavior do you expect?

tuloski, 2018-01-26:
Well, more options are always better.
Something like MAV_CMD_ABORT_RTL --> stops the mission and performs RTL
MAV_CMD_ABORT_STOP --> loiter in place
I'm saying random stuff, but it is weird to not have a mission abort at protocol level. The only way now is to change flight mode taking over with the RC (it's an "hard" abort). Also a PAUSE for the mission can be useful, where the mission can be recovered after some manual loitering.

magicrub, 2018-01-26:
>MAV_CMD_ABORT_RTL --> stops the mission and performs RTL

What's wrong with the current method of switching to RTL?

>MAV_CMD_ABORT_STOP --> loiter in place

What's wrong with the current method of switching to Loiter?

At least on ArduPilot none of these are a hard abort. They can all be done without an RC and all work on car, plane, copters, and switching back to AUTO would resume the mission. Perhaps this is a limitation of the flight stack you're using?

tuloski, 2018-01-26:
I agree that you can achieve what I'm saying with current software, but mine was more a theoretical discussion at protocol level. For me having a command for abort mission is semantically different than achieving the same result switching to LOITER.

R-Lefebvre, 2018-01-26:
But if the abort mission command, simply puts the machine in Loiter mode, but it's the code on the aircraft that makes the switch in response to the command, rather than the GCS issuing the Loiter command, how is this better or different?

Also Tom, don't forget, if you actually want to Stop, and fast, switch to Brake/Stop mode.  It will stop much faster than Loiter.

DonLakeFlyer, 2018-01-29:
The problem is that flight modes are a black box to a ground station. Unless you know specific details about the firmware implemenation a ground station cannot change a flight mode to loiter for example.

Mission Start/Stop is just the tip of the iceberg with respect to this sort of generic protocol problem. Things like supporting guided style commands are other examples.

amilcarlucas, 2018-03-08:
Ardupilot flight modes are no longer a black box. They have enums now.

tuloski, 2018-03-08:
> Ardupilot flight modes are no longer a black box. They have enums now.

What does the enum mean?
Still the GS doesn't know how flight modes work and how they are implemented on the UAVs.

amilcarlucas, 2018-03-09:
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L922
Those are the enums, their meaning is implicit in their name.

hamishwillee, 2020-04-23:
Closing. If this is a problem still (and I don't think it is) then reopen and I will look closer.

closed 2020-04-23 by hamishwillee