mavlink/mavlink #831: HIGH_LATENCY2 review and finalize

dagar, 2018-01-29:


DonLakeFlyer, 2018-01-31:
I have the QGC side mostly implemented now. Here are the problems I ran into:
* flight_mode - This seems really strange. There is now a firmware specific enum you have to deal with. Makes it a royal pain to implement. Which ones of the enum are PX4 and whichs ones are the ArduPilot flight modes? I need to be able to translate from these enum values back to base_mode and custom_mode internal to QGC. What would be better is to just send two bytes of the existing custom_mode values. Both PX4 and ArduPilot only needs two bytes of info. And base_mode is not really used other than to say just use custom_mode (MAV_MODE_FLAG_CUSTOM_MODE_ENABLED). Still strangely firmware specific, but less so. And not a nightmare to implement/maintain since there is only single place in GCS code which deal with flight modes based on base_mode/custom_mode. Instead of two places in code required different implementations. Both which needs to be kept up to date as things change.
* eph/epv - I assume UINT8_MAX means unknown. Just like GPS_RAW_INT? If so, comment text needs to be updated. If not, what is the indicators for unknown.
* battery - DNU? I assume that means unknown? SYS_STATUS says "-1: autopilot estimate the remaining battery". Which is also a meaningless statement. If -1 means unknown, then both of these messages should be updated to reflect that.
* There is no way to know if the vehicle is armed, but I guess that is ok.

Note: I still have not looked at implementing failsage and failure_flags. Once I'm done with that I may have more comments on those.

acfloria, 2018-02-01:
- flight_mode  
I also thought about that a bit more while I implemented it on the PX4 side. In my opinion it would be a waste to use 2 bytes to transmit the flight mode as it can be done more efficient. But I agree having a separate enum for PX4 and ArduPilot is not ideal. So I have two ideas:
    1. Due to the latency and running costs of the link it should not be used while in a mode which requires RC inputs. This assumption would allow for a simplified combined definition:
        - `FLIGHT_MODE_RC`: Any mode which requires 
        - `FLIGHT_MODE_RC`: 
    2. Is it possible to reduce the size of the of the `custom_mode` in general to 1 byte. I would say that 256 different flight modes should be sufficient. This would probably be a cleaner solution but would require synchronized work on the PX4, ArduPilot, and QGC side and possibly also influences other autopilots.

 - eph/epv  
That is how I implemented it for PX4. I think that this definition would be reasonable.

- battery  
Yes, -1 means do not use or invalid.

 - armed  
As the high latency link is a fallback from normal telemetry the basic assumption is that the vehicle is flying and armed when the high latency message is received. As each message will cost the autopilot should not send a message while disarmed and not flying.

- failsafe/failure flags  
I think that we could combine them to one `uint16_t`. I would suggest that the freed byte is defined as a `custom_flags` field.

acfloria, 2018-02-01:
- flight_mode  
I also thought about that a bit more while I implemented it on the PX4 side. In my opinion it would be a waste to use 2 bytes to transmit the flight mode as it can be done more efficient. But I agree having a separate enum for PX4 and ArduPilot is not ideal. So I have two ideas:
    1. This message is used when the plane is outside the regular telemetry range which also most probably means outside the RC range. This means that it should not be used while in a mode which requires RC inputs. This assumption would allow for a simplified combined definition of only the auto modes:
        - `FLIGHT_MODE_RC`: Any mode which requires rc input. In QGC an error message should be displayed if this mode is active.
        - `FLIGHT_MODE_AUTO`: Also known as mission, simple waypoint following
        - `FLIGHT_MODE_GUIDED`: Only the goal is set by the user, intermediate waypoints are planned on the autopilot 
        - `FLIGHT_MODE_LAND`: Any landing mode
        - `FLIGHT_MODE_TAKEOFF`: Any takeoff mode
        - `FLIGHT_MODE_RTL`: Any return to launch mode
        - `FLIGHT_MODE_RTGS`: Any return to ground station mode
        - `FLIGHT_MODE_LOITER`: Standard loitering
        - `FLIGHT_MODE_CIRCLE`: Loitering without global position tracking
        - `FLIGHT_MODE_AAT`: Avoid air traffic
        - `FLIGHT_MODE_FAILSAFE`: Any failsafe mode not represented by any of the previous defined modes.

    Would such a definition of the flight mode also be able to represent all the different auto flight modes of ArduPilot? @DonLakeFlyer would definition be ok to implement on the QGC side?

    2. Is it possible to reduce the size of the of the `custom_mode` in general to 1 byte? My gut feeling says that 256 different flight modes should be sufficient. This would probably be a cleaner solution but would require large changes to different code bases and synchronized work on the PX4, ArduPilot, and QGC side and possibly also influences other autopilots. That is why I would opt for the other solution.

 - eph/epv  
That is how I implemented it for PX4. I think that this definition would be reasonable.

- battery  
Yes, -1 means do not use or invalid.

 - armed  
As the high latency link is a fallback from normal telemetry the basic assumption is that the vehicle is flying and armed when the high latency message is received. As each message will cost the autopilot should not send a message while disarmed and not flying.

- failsafe/failure flags  
I think that we could combine them to one `uint16_t`. I would suggest that the freed byte is defined as a `custom_flags` field.

DonLakeFlyer, 2018-02-01:
I just don't think it makes sense for normal flight mode handling to be opaque (base_mode + custom_mode) and then suddenly with HIGH_LATENCY2 they are not opaque are you need a whole different system to handle them. Not to mention that with that there are now two different sets of flight modes. Is a ground station supposed to try to show different flight modes based on latency? Doesn't make sense to me. How do you show that to the user. Flight modes should be handled in one way no matter what.

dagar, 2018-02-01:
I completely agree that they should be handled in one way, but I consider opaque flight mode handling to be a severe flaw with how mavlink is commonly used. 
The spec could easily contain common modes (covering the majority) and even all project specializations in a fraction of a single byte. Instead we waste 5 bytes (+1 byte base_mode) and it all falls outside of the spec? 

Short term I could live with HIGH_LATENCY2 using a 2 byte custom_mode that works with the existing custom_mode, but longer term I think it's pathetic if we can't find a way for something this fundamental to be part of the spec.

DonLakeFlyer, 2018-02-01:
Can someone tell me what standard mavlink messages ```failure_flags``` and  ```failsafe``` relate to? Trying to figure out if QGC uses these.

acfloria, 2018-02-02:
I could live with a 2 byte custom_mode for now if we fuse the failsafe flags but also think we should reduce it to 1 byte for the future.

> Can someone tell me what standard mavlink messages failure_flags and failsafe relate to? Trying to figure out if QGC uses these.

- The sensor related flags (gps, airspeed, baro, accel, gyro, mag, lidar, geofence, rc) relate to MAV_SYS_STATUS_SENSOR flags. Though for example the there is only one accel reported.

- MAV_FAILSAFE_FLAG_TELEM_LOSS could be removed as it does actually make no sense because we cannot report it anyway.

- MAV_FAILSAFE_FLAG_ENGINE: I couldn't find any other Mavlink message which would report that.

- MAV_FAILSAFE_FLAG_BATTERY: This would relate the best to the actual battery voltage. The flag is true if the vehicle is failsafe mode due to low battery voltage.

- MAV_FAILURE_FLAG_MISSION: This flag is true if executing the current mission failed. I didn't find a similar Mavlink message.

- MAV_FAILURE_FLAG_ESTIMATOR: This flag reports if there is a failure in the estimator. I didn't find a similar Mavlink message.

- MAV_FAILURE_FLAG_OFFBOARD_LINK: This flag is true if the offboard link timed out. I didn't find a similar Mavlink message.

DonLakeFlyer, 2018-02-02:
So thats HIGH_LATENCY2.failure. What mavlink message does HIGH_LATENCY2.failsafe correspond to?

DonLakeFlyer, 2018-02-03:
More questions/review here: https://github.com/mavlink/qgroundcontrol/pull/6098

amilcarlucas, 2018-02-04:
Please consider this PR as well https://github.com/mavlink/mavlink/pull/826

acfloria, 2018-02-05:
@DonLakeFlyer I answer here the questions asked in [mavlink/qgroundcontrol#6098](https://github.com/mavlink/qgroundcontrol/pull/6098)
- Some of the failsafe flags also correspond to MAV_SYS_STATUS_SENSOR but most of them are -
currently not represented. I would suggest that we merge failsafe and failure anyway in a revision of the high latency message.

- I agree, that the naming can be improved for the flags.

- The reason why some of the flags are currently not represented by QGC is that we want to avoid transmitting the STATUS_TEXT. So we tried to gather them in the additional flags. For QGC I planned anyway to add a high latency widget to display those flags.

DonLakeFlyer, 2018-02-05:
Ok, understood.

> For QGC I planned anyway to add a high latency widget to display those flags.

The sensor health flags already display in the Health page of the intrument panel. So these new ones can go there as well.

dagar, 2018-02-08:
I think we should consider replacing the original HIGH_LATENCY with HIGH_LATENCY2. If both major mavlink adopters signoff on it (and we know the original is barely used if at all) I don't see why it isn't okay.

WickedShell, 2018-02-08:
@dagar no, it utterly breaks the concept of MAVLink as a write only protocol, and you don't know who else might have grabbed the definition file in the interim. Replacing old messages will only fracture the user base.

DonLakeFlyer, 2018-02-08:
To me the important rule here is don't spec things in the protocol that nobody (PX4/ArduPilot) is using yet. Because once you actually use them they almost always need to be changed. Then you get stuck with leftover crap in the protocol.

dagar, 2018-02-08:
Yes definitely. We should discuss and layout some basic ground rules for what to do. Either marking things DRAFT and allowing them to change, or improving the development workflow so feature branches are easily usable.

amilcarlucas, 2018-02-08:
@Dagar Currently stuff is not marked as Draft. The only marker in use is WIP.
I think feature branches are the way to go too. Everybody can merge them together locally and create their own local mavlink flavor. While keeping the public master clean and stable.

dagar, 2018-02-08:
I didn't realize HIGH_LATENCY2 dropped the marker. I think we should definitely keep it until we've tested it fully. @DonLakeFlyer can we readd either DRAFT or WIP right away? I don't care which.

The problem is many projects include the c_library mavlink repository (https://github.com/mavlink/c_library_v2) directly instead of the message definitions. So it's more of a pain to test everything together until it makes it into the actual mavlink library they use. One option might be if feature branches in mavlink generate a corresponding branch in c_library_v2.

DonLakeFlyer, 2018-02-09:
Fine by me. I'm not really involved in the actuall mavlink pulls for this.

julianoes, 2019-05-08:
HIGH_LATENCY does not seem to be WIP anymore, so I'm closing this.

closed 2019-05-08 by julianoes