mavlink/mavlink #832: Remove v0.9 from mavlink message_definitions?

samacumen, 2018-01-29:
I notice that v0.9 is hardly used. In QGC and PX4, v1.0 is used. Why can't we remove v0,9 from message definitions? IMHO, it does a world a lot more good.

Also, I don't really get the meaning of protocol versions Mavlink v1.0, v2.0 (packet format) versus v0.9 and v1.0 under message definitions. Which maps to which ones? Can someone explain me the difference between these? Why are there a different nomenclature in version numbering?

hamishwillee, 2018-01-30:
>  Why can't we remove v0,9 from message definitions? IMHO, it does a world a lot more good.

Eventually. We tried this and found there were a lot of dependencies in the test code. Yes it could be done, but not really worth the effort.

> Also, I don't really get the meaning of protocol versions Mavlink v1.0, v2.0 (packet format) versus v0.9 and v1.0 under message definitions. Which maps to which ones? Can someone explain me the difference between these? Why are there a different nomenclature in version numbering?

I believe that the message definitions actually changed between v0.9 and v1.0. The definitions for MAVLink 2.0 were changed in a backwards compatible way with 1.0 so there was no need to create a new tree. The current protocol version is listed in the docs https://mavlink.io/en/messages/common.html#mavlink-protocol-version

I do agree this is confusing - we probably should at least have a doc that explains this. 

@julianoes What is the most pragmatic way to help people understand they can ignore 0.9 message definitions and that the 1.0 tree of message definitions covers both v1.0 and 2.0? Options:
- Delete the 0.9 tree
- Add readme in tree explaining it is deprecated (and also in 1.0 tree explaining covers both)
- Add note [here](https://mavlink.io/en/mavlink_version.html#mavlink-version) explaining there is  legacy beta version 0.9 that can be ignored.
- Other?

julianoes, 2018-01-31:
@hamishwillee ok, just try a PR where you delete it and see how it goes in CI, etc. And then we fix it from there.

amilcarlucas, 2018-03-08:
I think this can be closed and the discussion can be continued on that PR

closed 2018-03-08 by LorenzMeier