mavlink/mavlink #834: Architectural considerations: The CRC extra byte and intermediate systems

rodizio1, 2018-02-06:
While running into issues with a kind of mavlink-forwarder/proxy (or whatever you might want to call it) I was working on lately, I stumbled across that extra CRC byte. Having read the article on mavlink.org about it, I understand it's purpose is to make sure that messages with incompatible fields in the payload do not get accidentally (and wrongly) decoded. In theory a very good idea, however, the way this has been solved has created a lot of issues. This seems to be unchanged with Mavlink v2 (and even made worse due to the added complexity of the flag_unlocked_crc thing).

Why is it bad?

Without deep knowledge about the payload, there is no way to reliably determine where the start and end of a message is and if the message is valid:

- It forces intermediate systems to have knowledge about the payload and all the different implementations
- Thus, these systems become unnecessarily more complex, more error-prone, incompatible
- There is no way of actually knowing what happened. The code calculates the CRC, in case it's wrong, it has no way of telling if garbage was recived or just a message with a different extra byte
- For the users, obviously, it's the same. They just see dropped messages and start wondering about RF-link problems or electromagnetic fields flipping bits on the serial line ...

I realize I'm quite late to the party, but can you guys please consider following a more layered-approach? Think of routers on the Internet. They don't need to have knowledge about the upper layers, all they need to do is look at the IP header and forward packets based on the information found there. It doesn't matter if the TCP protocol gets changed, they will still forward everything just fine. Or think of the HTTP protocol, it doesn't matter if some new header fields get defined, as long as the transmission follows the rules of the HTTP protocol, every proxy or loadbalancer will work just fine. Also, the TCP checksum still works fine. If routers would be 'randomly' (atleast that's what it looks like to the user) dropping packets depending on what particular implementation detail of TCP or HTTP may be different with different webservers and routers, I think the Internet simply wouldn't exist as it does today.

As it is now, there is no way anybody could ever create an intermediate system that can deal with all message-types/flightcontrols/GCS-softwares and that 'just works'.

Even if all projects kept their mavlink libraries perfectly up-to-date and in-sync, there are still incompatibilities, and as far as I've understood, there is no way to link the C library with a specific dialect. Even if linking the C library with a specific dialect becomes possible, I still don't see a convenient and not error-prone way (both for the programmer and the user) to use it. The programmer would have to deal with a lot of dialects and make them configurable, the user would have to configure it. Users owning two or more different flightcontrol stacks have to re-configure everytime they uses a different plane/FC. Simply forgetting to configure the correct dialect may become very annoying when realizing mid-flight that whatever function doesn't work because the messages get dropped.

hamishwillee, 2018-02-06:
@rodizio1 I'm not the right person to discuss this with, but a few questions to help someone more knowledgable:
- Do you mean article on https://mavlink.io/en/ ? Specifically, the latest version is here: https://mavlink.io/en/serialization.html#field-reordering-and-crc-extra-calculation
- There is also some additional background on MAVLink 2 implementation here: https://github.com/mavlink/mavlink/blob/master/doc/MAVLink2.md and in its linked document.

rodizio1, 2018-02-06:
Yes, exactly those (and the one linked in the 2nd article: https://docs.google.com/document/d/1XtbD0ORNkhZ8eKrsbSIZNLyg9sFRXMXbsR2mp37KbIg/edit?usp=sharing)

hamishwillee, 2020-04-24:
I'm closing this as "mostly" invalid - specifically the CRC is not a problem, but the generator implementation may be. 

The CRC indicates that the sender and receiver have the same definition of the message at compile time. If they don't, the message is dropped, which makes sense because the receiver cannot process it. The failure may be because the message is either corrupted or incompatible - you can't resolve which. 

> It forces intermediate systems to have knowledge about the payload and all the different implementations

No it doesn't. The specification states that a system that processes a message must have a matching build of the library and that a system that routes a message should do so whether or not it understands the message. So a routing system should not need to know anything about the message.

The problem here is the implementation of the generated libraries - they are dropping the message if they can't process it. There are various issues on this topic in the Ardupilot/pymavlink generator repo, and some routing software handles this case correctly. 

> Thus, these systems become unnecessarily more complex, more error-prone, incompatible
There is no way of actually knowing what happened. The code calculates the CRC, in case it's wrong, it has no way of telling if garbage was recived or just a message with a different extra byte
> For the users, obviously, it's the same. They just see dropped messages and start wondering about RF-link problems or electromagnetic fields flipping bits on the serial line ...

No it isn't more complex or error prone, yes it is true that you can't tell whether the cause is corrupt or incompatible message. Yes a user might worry about this, but the fix is the routing software, not the spec.

closed 2020-04-24 by hamishwillee