mavlink/mavlink #838: Need help with the MAV_CMD_OVERRIDE_GOTO command

RohanKurane, 2018-02-11:
@LorenzMeier @lucasdemarchi @DonLakeFlyer 
I came across the MAV_CMD_OVERRIDE_GOTO command but that does not seem to change to HOLD mode in QGC when I send the following command:
cmd.command = MAV_CMD_OVERRIDE_GOTO;
cmd.target_system = TARGET_SYSTEM_ID;
cmd.target_component = MAV_COMP_ID_ALL;
cmd.param1 = MAV_GOTO_DO_HOLD;
cmd.param2 = MAV_GOTO_HOLD_AT_SPECIFIED_POSITION;
cmd.param3 = 0;
cmd.param4 = 0;
cmd.param5 = 0;
cmd.param6 = 0;
cmd.param7 = 10;
mavlink_msg_command_long_encode(255, MAV_COMP_ID_ALL, &msg, &cmd);
len = mavlink_msg_to_send_buffer((uint8_t *)buf, &msg);
r = sendto(tcp_fd, buf, len, 0, (struct sockaddr *)&sockaddr, sizeof(sockaddr));

What am I missing here to be able to switch to HOLD mode in QGC from my application ?

I am using QGC v3.2.7. 

I want to be able to send the command and change to HOLD mode. This should take the RTF to the specified altitude and hover there till it is landed. Please let me know how I can do this.

RohanKurane, 2018-02-12:
Also, can you tell me the right way to send a Takeoff command from an application? The drone crashes on a takeoff command. It does not hold in air when the mode is set to HOLD and a takeoff is done.

RohanKurane, 2018-02-12:
If I look in the QGC Console logs , when I takeoff using QGC it shows me the following fields
Rating: great
Wind: 0

But when I takeoff using my application code, it does not. So it seems like it crashes right away.

What would you need to help me debug this ?

RohanKurane, 2018-02-12:
I have uploaded a file with the QGC takeoff (qgc_takeoff.ulg). I was not able to get any logs when I did a takeoff from within my application. Why is that ? Does my application need to do logging like QGC ?

RohanKurane, 2018-02-12:
Sorry to bug. But this is something I need to get working today. Any help would be really appreciated. 
Basically in short, the MAV_CMD_NAV_TAKEOFF command does not work the expected way from my application. It changes the mode to 'takeoff' correctly but does not climb to the altitude (specified in cmd.param7). Instead it takes off in a random direction and crashes. 

When I do the same thru QGC, it works. I have uploaded the PX4 log (gc_takeoff.ulg) on https://logs.px4.io/  for the QGC scenario. Please let me know if there is anything else I can provide to help you diagnose this.

hamishwillee, 2018-02-12:
@RohanKurane YOu might want to see if anything can be learned from the Dronecore source code - it implements the MAVLink takeoff command here: https://github.com/dronecore/DroneCore/blob/develop/plugins/action/action_impl.cpp#L102

zehortigoza, 2018-02-13:
Not sure if this is the problem but you should not use the same system ID as GCS/QGC(255).

RohanKurane, 2018-02-13:
It works for all other commands I send thru the application. What system ID should I use ?

zehortigoza, 2018-02-13:
Any not used system id in your setup.

Before ask PX4 to arm and then take off you should also check the PX4 state.
it have GPS lock and so on...? https://github.com/intel/mavlink-router/blob/master/examples/px4-offboard-mode.cpp#L395

Another tip is use PX4 simulators to test your application before the real deal.

RohanKurane, 2018-02-13:
I can change the to use 200 in the command
mavlink_msg_command_long_encode(200, MAV_COMP_ID_ALL, &msg, &cmd); to arm and takeoff

As I mentioned, it works fine with QGC. I do the same exact steps.  And it works. 
Set to HOLD mode using QGC
Arm using QGC
Takeoff using QGC
Return using QGC

Then I do this - 
Set tio HOLD mode using QGC
Arm using my application  (arm is successful)
Takeoff using my application (crashes) 

A side question - What is the MAV_CMD to set to HOLD mode ?  Like Land mode has MAV_CMD_NAV_LAND.

amilcarlucas, 2018-02-20:
IMHO this is not a mavlink issue. This is a PX4 issue, and should be moved to the PX4 repository.

RohanKurane, 2018-02-20:
@amilcarlucas  What are the mavlink commands to enter HOLD state ?

amilcarlucas, 2018-02-21:
`MAV_CMD_DO_SET_MODE` and please move this to the PX4 support forum

closed 2018-02-21 by LorenzMeier