mavlink/mavlink #840: Can't successfully clone MAVLink with submodules

hamishwillee, 2018-02-15:
Following the instructions fails with pymavlink fetch:
```
>git clone https://github.com/mavlink/mavlink.git --recursive
Cloning into 'mavlink'...
remote: Counting objects: 14940, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 14940 (delta 3), reused 9 (delta 3), pack-reused 14929
Receiving objects: 100% (14940/14940), 6.76 MiB | 267.00 KiB/s, done.
Resolving deltas: 100% (9860/9860), done.
Submodule 'pymavlink' (https://github.com/mavlink/pymavlink.git) registered for path 'pymavlink'
Cloning into 'D:/Github/forks/mavlink/test/mavlink/pymavlink'...
error: no such remote ref 5f31f68e6e373336df12d1cea31626097cece6bc
Fetched in submodule path 'pymavlink', but it did not contain 5f31f68e6e373336df12d1cea31626097cece6bc. Direct fetching of that commit failed.
```

Same problem (on update) if you use:
git clone https://github.com/mavlink/mavlink.git
git submodule init
git submodule update
```

magicrub, 2018-02-15:
Have you tried
git submodule update --recursive

hamishwillee, 2018-02-15:
Yes, I have tried this (and other variants below) on both windows and Linux. I am pretty sure there is actually something wrong with the submodule reference. Thoughts?

Using  `git submodule update --recursive` :
```
D:\Github\forks\mavlink\test4\mavlink>git submodule init
Submodule 'pymavlink' (https://github.com/mavlink/pymavlink.git) registered for path 'pymavlink'

D:\Github\forks\mavlink\test4\mavlink>git submodule update --recursive
Cloning into 'D:/Github/forks/mavlink/test4/mavlink/pymavlink'...
error: no such remote ref 5f31f68e6e373336df12d1cea31626097cece6bc
Fetched in submodule path 'pymavlink', but it did not contain 5f31f68e6e373336df12d1cea31626097cece6bc. Direct fetching of that commit failed.
```
And `git submodule update --init --recursive`:
```
D:\Github\forks\mavlink\test2\mavlink>git submodule update --init --recursive
Submodule 'pymavlink' (https://github.com/mavlink/pymavlink.git) registered for path 'pymavlink'
Cloning into 'D:/Github/forks/mavlink/test2/mavlink/pymavlink'...
error: no such remote ref 5f31f68e6e373336df12d1cea31626097cece6bc
Fetched in submodule path 'pymavlink', but it did not contain 5f31f68e6e373336df12d1cea31626097cece6bc. Direct fetching of that commit failed.
```

assafstoler, 2018-02-16:
Issue is with one of the last two commits. if you don't need latest (and until it's fixed) you can go back a couple commits and all will work by:
git reset --hard  44726fe1e8932cb181d4923f407664024cc3f4df
git submodule init
git submodule update

hamishwillee, 2018-02-16:
Thanks @assafstoler - thank you very much for that! Do you know who is working on the issue?

assafstoler, 2018-02-16:
@hamishwillee Welcome, I am not a contributor to this project, but looking at the Git history It appears the offending commit is 2e7281d1, by "Dr.-Ing. Amilcar Do Carmo Lucas <amilcar.lucas@iav.de>".I bet my money on him forgetting to push his updated sub-module together with the ref. update.

hamishwillee, 2018-02-16:
Thanks @assafstoler - I'll chase him.

hamishwillee, 2018-02-16:
@amilcarlucas Can you please see this thread? You may (or may not) be aware that mavlink isn't syncing the submodule properly - can you look at this?

amilcarlucas, 2018-02-16:
Hi @hamishwillee,

I commented on https://github.com/mavlink/mavlink/pull/826#issuecomment-364033999 that mavlink/pymavlink#7 was required, before merging #826 
@dagar missed that comment and went in and merged #826 

I think I have fixed it now, by pushing into a new pymavlink branch. Please re-try.

dagar, 2018-02-16:
I'm not following the history here. Why did CI pass in https://github.com/mavlink/mavlink/pull/826?

amilcarlucas, 2018-02-19:
Problem should be solved now. this can be closed

hamishwillee, 2018-02-19:
Yes it is. Thanks @amilcarlucas

closed 2018-02-19 by hamishwillee