mavlink/mavlink #849: CRC calculation in Java library, V2 protocol

rcuadradosa, 2018-02-21:
I have just upgraded my software for using Mavlink V2 protocol but I have noticed that several packets are discarded after checking the CRC, and all those packets are compressed by the trimPayload() method.
The `generateCRC(int payloadSize)`, method use the variable **len** to update the checksum
`private void generateCRC(int payloadSize){ ... crc.update_checksum(len); ....}`

Debugging, I have detected that the value of **len** (field of MAVLinkPacket) is different in the encoding and decoding phase. 
In the encoding phase, **len** is initialized with **MAVLINK_MSG_LENGTH** value (the maximum value of the message), but the Parser class uses the second byte of the data packet to initialize the MAVLinkPacket and consequently to initialize the **len** field. If the payload ends with 0 value bytes, the payload is trimmed and the length employed in the encoding CRC calculation is different from the value used in decoding CRC calculation.

Is this issue fixed in any branch? Am I missing something?

Thanks!

WickedShell, 2018-02-21:
@rcuadradosa I don't think you are missing anything. There are a number of outstanding bugs/bad behavior in the Java MAVLink 2 stuff that have come, and this looks to be an error (although should be a straightforward one to correct).

amilcarlucas, 2018-02-21:
Yes, it might be a bug, but it is an upstream bug in https://github.com/ArduPilot/pymavlink/issues
and does not belong here. Please re-post it there.

AFAIK this can be closed now

closed 2018-02-21 by LorenzMeier

WickedShell, 2018-02-21:
@amilcarlucas No it's not an upstream bug. I refused the Java changes as they are bad code, so the MAVLink/pymavlink is the only location where the Java parser behaves this way, and you can't open issues on MAVLink/pymavlink, so this was opened in the correct location.

@LorenzMeier this should be reopened.