mavlink/mavlink #850: Links do not exist anymore. please correct.

samacumen, 2018-02-25:
The link in some of the readme pages does not work.

e.g. 
1) on https://github.com/mavlink/mavlink/tree/master/examples/linux (no quickstart guide at this link, http://qgroundcontrol.org/dev/mavlink_linux_integration_tutorial)

2) message_definitions/v1.0/common.xml (http://qgroundcontrol.org/parameter_interface)

Please correct these and also in http://qgroundcontrol.org/mavlink/start many links lead no-where.

hamishwillee, 2018-02-25:
@samacumen Thanks. 

>  https://github.com/mavlink/mavlink/tree/master/examples/linux (no quickstart guide at this link, http://qgroundcontrol.org/dev/mavlink_linux_integration_tutorial)

I have simply deleted the link from the document - there is no replacement: https://github.com/mavlink/mavlink/pull/851

> message_definitions/v1.0/common.xml (http://qgroundcontrol.org/parameter_interface)

I have fixed up all inks to point to the correct new site/documents
https://github.com/mavlink/mavlink/pull/852

> Please correct these and also in http://qgroundcontrol.org/mavlink/start many links lead no-where.

The old site on qgroundcontrol.org is dead - as it states on the home link up the top. The old site still has some old/broken links, but again, these should be ignored while we migrate materials (as it says up the top of the home page). The new site is: https://mavlink.io/en/

I'll let this closed, but the PRs should fix the problems in the coming days.

closed 2018-02-25 by hamishwillee

samacumen, 2018-02-25:
Perfect. Thanks for your time on this.