mavlink/mavlink #857: Feature request

Stifael, 2018-03-12:
Currently MAV_CMD_DO_CHANGE_SPEED (http://mavlink.org/messages/common#ATT_POS_MOCAP) only allows to change speed horizontally. I would like to define one of the empty slots as a speed change command for vertical direction.

Stifael, 2018-03-12:
Another option would be to add a new type: 0 = Air-speed, 1 = Ground-speed, 2 = Vertical-speed

amilcarlucas, 2018-03-12:
A concrete Pull request is a lot better than an issue.
Can you open a pull request ?

Stifael, 2018-03-12:
Well, I actually was hoping to get some feedback first

WickedShell, 2018-03-12:
Extending the mapping on param1 makes more sense then consuming a new field. (As on a new field you would also have to handle the fact that old GCS's might send it to you with 0 because they didn't know it should be something else)

DonLakeFlyer, 2018-03-12:
> As on a new field you would also have to handle the fact that old GCS's might send it to you with 0 because they didn't know it should be something else

FYI: That whole mess really needs to be standardized. The way it currently stands, anything marked as "reserved" is unusable in the future and can never be change without possible issues. Reserved really needs to be specifically set to NaN in order to make them future usable for anything.

So the only way to update MAV_CMD_DO_CHANGE_SPEED is to add to param 1. Would be good to have an enum as well.

hamishwillee, 2020-04-23:
I'm assuming this either happened, or is not needed. Closing. Reopen if that assumption is invalid :-)

closed 2020-04-23 by hamishwillee