mavlink/mavlink #859: Support for Video streaming messages

shakthi-prashanth-m, 2018-03-13:
Below MAVLink messages related to Video streaming are WIP:

- [MAV_CMD_VIDEO_START_STREAMING](http://mavlink.org/messages/common/#MAV_CMD_VIDEO_START_STREAMING)
- [MAV_CMD_VIDEO_STOP_STREAMING](http://mavlink.org/messages/common/#MAV_CMD_VIDEO_STOP_STREAMING)
- [MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION](http://mavlink.org/messages/common/#MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION)

I am working on developing [DroneCore camera plugin](https://github.com/dronecore/DroneCore/issues/292) to implement [MAVLink Camera Protocol](https://mavlink.io/en/protocol/camera.html) along with Lalit Begani (lalit.kumar.begani@intel.com) who is contributor to [Camera Streaming Daemon](https://github.com/intel/camera-streaming-daemon). So, DroneCore camera plugin should implement video streaming as well.

Are these messages obsolete ? OR Is it WIP indeed ?

hamishwillee, 2018-03-13:
Hi Shakthi,

My _understanding_ is that they are WIP because there is nothing out there that tests them and so no validation that they meet real use cases - CSD does not support them yet and there may be some need for support in firmware.

@lbegani is the right person to talk to about what is needed to bring these to life.

shakthi-prashanth-m, 2018-03-13:
Yeah, I spoke @lbegani. 
>MY (unconfirmed) understanding is that right now the CSD supports streaming

Yes. Now, is it worthwhile to handle these video streaming messages in CSD ? Camera plugin would request streaming URL using `MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION`. As they're marked as WIP and also it is _NOT_ part of MAVLink Camera Protocol! So, Me & @lbegani worried whether these messages are obsolete.

shakthi-prashanth-m, 2018-03-13:
I missed two of the videos streaming commands & messages. So, complete list is here:

- [MAV_CMD_VIDEO_START_STREAMING](http://mavlink.org/messages/common/#MAV_CMD_VIDEO_START_STREAMING)
- [MAV_CMD_VIDEO_STOP_STREAMING](http://mavlink.org/messages/common/#MAV_CMD_VIDEO_STOP_STREAMING)
- [MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION](http://mavlink.org/messages/common/#MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION)
- [VIDEO_STREAM_INFORMATION](http://mavlink.org/messages/common/#VIDEO_STREAM_INFORMATION
)
- [SET_VIDEO_STREAM_SETTINGS](http://mavlink.org/messages/common/#SET_VIDEO_STREAM_SETTINGS)

hamishwillee, 2018-03-13:
So IMO streaming is an important use case that it makes sense to drive from MAVLink. They aren't obsolete unless they have been superceded by something else that does the same thing instead (?). It may well be that they are "planned" to be part of the Camera Protocol, but simply didn't become part of that yet "while" they are WIP.

@dogmaphobic Can you shed any light on the [video streaming messages](https://github.com/mavlink/mavlink/issues/859#issuecomment-372847891) and the MAVLink camera protocol?

dogmaphobic, 2018-03-14:
Yes, it is in my todo list. I'm planning on implementing this in Gazebo. I already added the camera (start/stop image/video) capture API. This would be next.

hamishwillee, 2018-03-14:
Thanks @dogmaphobic ! 

@shakthi-prashanth-m @lbegani So yes, you should think about implementing this. Being aware that until the WIP is gone and the messages are in the spec, they are subject to change.

shakthi-prashanth-m, 2018-03-14:
OK, Thanks @dogmaphobic for the info.
Thanks @hamishwillee for clarifying it. I shall discuss with @lbegani further.

vigouredelaruse, 2018-05-30:
hello - I also have an interest in what I think you're discussing -

essentially I see a use case for raspberry pi (camera) based situations (such as Navio2 which I have) 

if I follow you correctly, this idea would support shoehorning video data regardless of the link interface, such as over a telemetry radio. 

so though you wouldn't hope to achieve high resolution / high video throughput, one can at least see partitioning a 57600 baud UART link into 2 channels, 1 priority MAVLINK channel and one for video

hamishwillee, 2018-05-31:
@shakthi-prashanth-m @dogmaphobic These messages can have WIP removed now right? I.e. there is now at least one implementation (camera manager). If not, what is required for a message to go from WIP to accepted?

hamishwillee, 2018-05-31:
@vigouredelaruse The main idea of the [MAVLink Camera API ](https://mavlink.io/en/protocol/camera.html) is to configure camera capture. The streaming here is via RTSP. 

As I understand it there is no particular thinking around shoehorning the streaming on a particular channel. The best place to discuss this would be the mavlink channel on slack.

lbegani, 2018-05-31:
@hamishwillee camera-manager is not handling video streaming messages yet.
@vigouredelaruse  - The idea of the video streaming messages under discussion is to start/stop/query/control video streaming which will happen over UDP or RTSP.

hamishwillee, 2018-05-31:
> camera-manager is not handling video streaming messages yet.

Thanks. I'd forgotten the [limitation on this page](https://hamishwillee.gitbooks.io/camera-manager/content/en/guide/overview.html): *At time of writing (March 2018) the camera protocol, and hence DCM, do not yet include a formal specification for managing or advertising RTSP video streams.*

When is that happening?

lbegani, 2018-06-01:
> When is that happening?

I do not have an ETA for this yet. I think the first step would be to work on the messages for video streaming (as they are outdated now) and take it from WIP to work done. I will discuss with @dogmaphobic  and @shakthi-prashanth-m to try to take this to conclusion.

jaxxzer, 2019-02-19:
Why are WIP messages in mavlink master?

hamishwillee, 2019-02-19:
@jaxxzer The model is that if you have a message that belongs in master but you still want to be able to change it during implementation testing you declare it as WIP. Yes, they still are in WIP.

@dogmaphobic You were doing implementation work with these - is there a plan to move these out of WIP any time soon?

dogmaphobic, 2019-02-20:
There are several WIPs on things I'm not working on. Those would have to be explained by the ones implementing it. Generally, a proposal is done and discussed. Once an agreement is made, the command/message is added to master but in a _WIP_ state until actual development is done (both on the vehicle/camera side and ground station). Only once it is known that things work as they should we removed the _WIP_ state. This is because no matter how much planning you put ahead of time, you invariably bump into surprises when it is actually deployed and used.

My WIPs:
* MAV_CMD_SET_CAMERA_ZOOM
* MAV_CMD_SET_CAMERA_FOCUS
* MAV_CMD_VIDEO_START_STREAMING
* MAV_CMD_VIDEO_STOP_STREAMING
* MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION
* MAV_CMD_REQUEST_VIDEO_STREAM_STATUS
* VIDEO_STREAM_INFORMATION
* VIDEO_STREAM_STATUS

These are all things that are still under development and only once I know for sure they all do what they are supposed to do and nothing is missing, will I remove the WIP state. That's usually the case when it makes it into QGC and shipped as "stable".

hamishwillee, 2020-04-24:
Closing, things have moved one.  The spec is finalised though some messages are still marked as WIP.

closed 2020-04-24 by hamishwillee