mavlink/mavlink #866: Difference between GLOBAL_POSITION_INT_COV and GLOBAL_POSITION_INT

alexx600, 2018-03-17:
I cant find detailed information about GLOBAL_POSITION_INT_COV  and GLOBAL_POSITION_INT frames.

I would like to ask if there is any difference in position (lat and long) in frames GLOBAL_POSITION_INT_COV and GLOBAL_POSITION_INT or difference is only in Covariance matrix?

TSC21, 2018-03-17:
Those are not frames, but rather Mavlink msg types. http://mavlink.org/messages/common#GLOBAL_POSITION_INT and http://mavlink.org/messages/common#GLOBAL_POSITION_INT_COV. And yes, besides one or two fields, the main difference is the covariance matrix.

hamishwillee, 2018-03-18:
@TSC21 Just FYI, can we start to point people towards mavlink.io instead of mavlink.org?
The mavlink.io is "better" because it is integrated with all the docs. The only difference is that the mavlink.io version may lag a change in an xml file by (max) 24 hours while mavlink.org is instantaneous. We will probably forward all links at some point.
 
https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT
https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT_COV

TSC21, 2018-03-20:
> @TSC21 Just FYI, can we start to point people towards mavlink.io instead of mavlink.org?

Sure thing!

amilcarlucas, 2018-04-12:
Can this be closed now ?

closed 2018-04-13 by mhkabir