mavlink/mavlink #868: Missions should support storage of unique identifier to be able to match a mission that was stored on a device even without permanent connection to the device

thomasgubler, 2018-03-19:
@dogmaphobic can you please fill in the details

dogmaphobic, 2018-03-19:
This needs coordination between @DonLakeFlyer and @dagar. 

There ought to be a way you can define a "mission id". You need to be able to map a given mission loaded on the vehicle to a mission stored on disk/cloud. The issue is that the only storage unit for missions is a command long (with 7 32-bit arguments) and in practicality, only the first 4 arguments can be used (the last 3 are hardcoded for coordinates). That gives us 4 32-bit arguments that could be concatenated into 16 bytes. Not much to guarantee a unique ID. 

How to solve this?

korigod, 2018-03-19:
What problem are you trying to solve? Do you need just some short identifier like mission hash to verify it? In general, the whole mission itself can be considered "mission identifier". Or do you need to store some metadata which you want to map to the given mission?

By the way, 128 bits are more than enough to guarantee a unique ID. UUIDs are 128-bit.

dogmaphobic, 2018-03-20:
>What problem are you trying to solve?

Flight plan. A "mission" is submitted as a flight plan to a (FAA type) authority using their schema. I need to match that flight plan ID with something within the mission.

>By the way, 128 bits are more than enough to guarantee a unique ID. UUIDs are 128-bit.

I stand corrected. I'm used to seeing it represented with digits like `123e4567-e89b-12d3-a456-426655440000` and didn't think about the absurdly large value stored within 128 bits.

thomasgubler, 2018-03-20:
➤ Gus Grubba commented:

>@dogmaphobic can you please fill in the details
What details and where? 
I'm still getting used to Wrike...

DonLakeFlyer, 2018-03-20:
The flight plan id you get back is 128 bits?

dogmaphobic, 2018-03-20:
No, it's an arbitrary string. There is no way we can fit that. Instead, we need to generate our own "unique" ID and store it with the mission. This same ID is then stored as metadata within the flight plan.

DonLakeFlyer, 2018-03-21:
The issue with needing to use all the bits from all the 7 param fields is I'm not sure what will happen to 5/6/7 when you are using MISSION_ITEM_INT format. I don't think all bits will round trip through that.

dogmaphobic, 2018-03-21:
Right, that's why I said we can only use the first four 32-bit parameters (128 bits).

DonLakeFlyer, 2018-03-21:
Sorry missed that. Duh. Then just creating a new mission command which is a no-op on the firmware side should work.

Adding @WickedShell

WickedShell, 2018-03-22:
What's the value in sticking it in the existing storage destination for mission commands? It can easily be done of course (and a no-op command is easy to handle), but I'm more curious why it should live in the mission, instead of being an attribute/separate fetch commands that the vehicle carries along with the mission? (Given that I can see you would want to look this up at any time, and wouldn't want to make assumptions about where the vehicle actually stored the command).

ArduPilot only dedicates 13 bytes of storage per mission item, which means we would need to entirely stride the item across 2 indexes, or reserve a dedicated spot for storing it (which I think is the preference and makes more sense then inserting the ID inside of the mission).

DonLakeFlyer, 2018-03-22:
> What's the value in sticking it in the existing storage destination for mission commands?

That's a really good point. I'm not sure it needs to actuall go into the mission itself.

DonLakeFlyer, 2018-03-22:
> Can a vehicle contain more than one mission?

Not currently. Although I would assume that is coming. I don't think I'd use a param. I'd just store it to the side of the mission data wherever that gets stored. With new mavlink commands to get/set.

DonLakeFlyer, 2018-03-23:
Any reason to not include a new message in the mission protocol. Something other than MISSION_ITEM(_INT)? Make a new MISSION_ID message set up the way you want. Protocol is to send this first then mission items. Then firmware implementations can store however/wherever they want. The id is tied to the mission. Firmware which don't understand this message will just ignore it.

WickedShell, 2018-03-23:
@DonLakeFlyer that would be my preference. It also seems to map out better as you can simply ask a vehicle for what the ID is without having to know where in a mission it was inserted, or what to do if someone inserts 2 waypoints with the same ID in a mission.

hamishwillee, 2020-04-23:
Keep alive. It is a good idea. Has been waiting on versioning mechanisms.