mavlink/mavlink #874: Storage & Image capture information messages are marked as WIP

shakthi-prashanth-m, 2018-03-26:
As discussed with @julianoes, MAVLink messages related to storage and image capture are marked as **WIP**; but they're being used by [DroneCore camera plugin](https://github.com/dronecore/DroneCore/pull/338/files/d6be938c26e0f31de9c57e37182997e397add0f7#diff-6029886b93401bf0054882a2cc199efdR32).

- http://mavlink.org/messages/common/#MAV_CMD_REQUEST_STORAGE_INFORMATION
- http://mavlink.org/messages/common/#STORAGE_INFORMATION
- http://mavlink.org/messages/common/#MAV_CMD_STORAGE_FORMAT
- http://mavlink.org/messages/common/#MAV_CMD_REQUEST_CAMERA_IMAGE_CAPTURE

If these messages works fine, then we need to remove "WIP".

julianoes, 2018-03-26:
And we should actually remove the storage ID sand use different component IDs in the same way as it is done for the cameras.

hamishwillee, 2019-02-06:
WIP has now been removed

closed 2019-02-06 by hamishwillee