mavlink/mavlink #881: MAVLink Generator should only display valid options for MAVLink 2 generation

hamishwillee, 2018-04-03:
MAVLink Generator GUI tool (mavgenerate.py) should only allow valid runtimes to be displayed as options if MAVLink 2 is selected as the protocol (Currently this is just C and C++ 11)

Something similar should be done with the underlying mavgen tool.

In the interim perhaps the limitation should be documented in https://mavlink.io/en/getting_started/generate_source.html

samacumen, 2018-04-03:
Thanks for reporting the issue @hamishwillee. I tried with protocol v2.0 and generated Python dependency correctly. For e.g. this screenshot below:

[image]
 
As you can see, it states headers (python lib) have been generated successfully. If it was never meant to be, strange how this issue got overlooked in last couple years.

hamishwillee, 2018-04-04:
@samacumen Did you actually check that the generated Python code supports MAVLink 2? The tool always reports that it was successful unless there is an error from the underlying mavgen. Both tools need to have checks to report errors for invalid combinations.

> If it was never meant to be, strange how this issue got overlooked in last couple years.

The vast majority of users use the pre-built C binaries rather than using the tool directly. Most of those users know what is supported and not supported. Would be great if the community could create MAVLink 2 support for those other runtimes, but in the meantime, just fixing the generator tools to fail for invalid cases is a good first step.

samacumen, 2018-04-04:
Thanks for this info @hamishwillee . I have always generated the C binaries of mavlink in the past. However, I wanted to use Python with mavlink v2.0, and was surprised (as you stated) that it is not yet supported. Today, most are using v2.0 and v1.0 is becoming more of a legacy.

I have for your reference, attached the mavgen generated python file with v1.0 and v2.0. With v2.0, I can see "MAVLink_setup_signing_message" messages included (which is part of v2.0 architecture) and generation with pack/unpack for WIRE_PROTOCOL_VERSION == '2.0'. However, the mavlink headers seem to look the same.

Could you have a quick look at your end on whether its inadvertently correct or its outright wrong generation?

[Generated mavgen python libs.zip](https://github.com/mavlink/mavlink/files/1874039/Generated.mavgen.python.libs.zip)

Thanks for the quick response.

hamishwillee, 2018-04-04:
Hi @julianoes 

As you may recall I recently updated docs with information that for MAVLink2 we only support C and C++ (https://github.com/mavlink/mavlink-devguide/pull/45).

@samacumen Says that Python appears to support MAVLink 2 too -https://github.com/mavlink/mavlink/issues/881#issuecomment-378451925

Can you please confirm what level of support the generated Python library has for MAVLink2 ?

vooon, 2018-04-04:
Python library **should** have full support, because mavproxy may use signing feature of v2.

samacumen, 2018-04-04:
I verified successful generation for all other languages as well. I compared v2.0 generation with its v1.0 counterpart, a quick look on generated headers seem as expected. What makes them non-supported then?

`MAVLink 2 source files can be generated for: 
 Python, Java, Javascript, C#, wlua, ObjC. 
`

> Please note: 
> C++ is only supported in wire protocol v2.0 but not in v1.0 (as has been documented correctly but hasn't been implemented as stated. Cpp headers for v1.0 are still generated, and user can overlook the pop-up warning. 

`_My $0.02: When user selects v1.0 in the mavgen tool, the 'Languages' not supported should completely disappear IMHO as it helps keep things clean and error free)._`

hamishwillee, 2018-04-11:
@julianoes I was informed that only C and C++ 11 generated libraries support MAVlink 2, but comments here seem to indicate that the other programming languages work too. Can you comment on what exactly is/is not supported so I can update the docs?

amilcarlucas, 2018-04-12:
Shouldn't this issue be reported upstream ? After all this is a pymavlink issue, right ?

hamishwillee, 2018-04-12:
@amilcarlucas No, at this point it is a "clarification" question - what programming languages are supported for mavlink2 - https://github.com/mavlink/mavlink/issues/881#issuecomment-380349852

Once that is answered then it might be that there is an upstream component required to fix it.

amilcarlucas, 2019-04-05:
AFAIK, pymavlink now properly does this. @samacumen can you retest with a recent pymavlink version ?

hamishwillee, 2019-05-03:
@amilcarlucas No. If it did then selecting C++11 in mavgenerate would not allow you to select 1.0 as shown below:

[image]

Thinking further about this there is no way to do this properly for mavgen (command line); it does the right thing by failing if it can't generate the relevant headers.

For mavgenerate we can do this by hardcoding the supported libraries in the UI. It's a fairly simple programming problem, just not a priority.