mavlink/mavlink #883: Video stream settings: Invalid values in the message elements

shakthi-prashanth-m, 2018-04-10:
In [MAVLink Video stream settings message](https://mavlink.io/en/messages/common.html#SET_VIDEO_STREAM_SETTINGS) below:

Field Name | Type | Description
-- | -- | --
target_system | uint8_t | system ID of the target
target_component | uint8_t | component ID of the target
camera_id | uint8_t | Camera ID (1 for first, 2 for second, etc.)
framerate | float | Frames per second (set to -1 for highest framerate possible)      (Units: Hz)
resolution_h | uint16_t | Resolution horizontal in pixels (set to -1 for highest resolution possible)      (Units: pix)
resolution_v | uint16_t | Resolution vertical in pixels (set to -1 for highest resolution possible)      (Units: pix)
bitrate | uint32_t | Bit rate in bits per second (set to -1 for auto)      (Units: bits/s)
rotation | uint16_t | Video image rotation clockwise (0-359 degrees)      (Units: deg)
uri | char[230] | Video stream URI

Default values for `resolution_h`, `resolution_v` and `bitrate` are -1! All these are of **unsigned type**. Hence it must have **zero** or similar to represent these default settings.

shakthi-prashanth-m, 2018-04-11:
Also, what is the use case of Camera client providing Video stream URI ? Anyway Camera server can start streaming on any port and provide the URI via `VIDEO_STREAM_INFORMATION`; isn't it ?

julianoes, 2018-04-23:
Good point. Also, note that `camera_id` was removed again in general, so it needs to be removed from this message as well, or made empty/reserved.

shakthi-prashanth-m, 2018-05-28:
PR #885 which fixes it, is in [pending ](https://github.com/mavlink/mavlink/pull/885#issuecomment-380792940) state. We can close this soon the discussion in its PR is concluded.

hamishwillee, 2019-02-06:
Closing as linked items are closed (ie decided not to keep messages)

closed 2019-02-06 by hamishwillee