mavlink/mavlink #886: Remove "No Action" field in several MAVLink commands

shakthi-prashanth-m, 2018-04-12:
### Consider below list of commands:

CMD ID | Field Name | Description
-- | -- | --
521 | MAV_CMD_REQUEST_CAMERA_INFORMATION | Request camera information (CAMERA_INFORMATION).
  | Mission Param #1 | 0: No action 1: Request camera capabilities
  | Mission Param #2 | Reserved (all remaining params)
522 | MAV_CMD_REQUEST_CAMERA_SETTINGS | Request camera settings (CAMERA_SETTINGS).
  | Mission Param #1 | 0: No Action 1: Request camera settings
  | Mission Param #2 | Reserved (all remaining params)
525 | MAV_CMD_REQUEST_STORAGE_INFORMATION | WIP:  Request storage information (STORAGE_INFORMATION). Use the command's  target_component to target a specific component's storage.
  | Mission Param #1 | Storage ID (0 for all, 1 for first, 2 for second, etc.)
  | Mission Param #2 | 0: No Action 1: Request storage information
  | Mission Param #3 | Reserved (all remaining params)
526 | MAV_CMD_STORAGE_FORMAT | WIP:  Format a storage medium. Once format is complete, a STORAGE_INFORMATION  message is sent. Use the command's target_component to target a  specific component's storage.
  | Mission Param #1 | Storage ID (1 for first, 2 for second, etc.)
  | Mission Param #2 | 0: No action 1: Format storage
  | Mission Param #3 | Reserved (all remaining params)
527 | MAV_CMD_REQUEST_CAMERA_CAPTURE_STATUS | Request camera capture status (CAMERA_CAPTURE_STATUS) 
  | Mission Param #1 | 0: No Action 1: Request camera capture status
  | Mission Param #2 | Reserved (all remaining params)
529 | MAV_CMD_RESET_CAMERA_SETTINGS | Reset all camera settings to Factory Default
  | Mission Param #1 | 0: No Action 1: Reset all settings
  | Mission Param #2 | Reserved (all remaining params)
2504 | MAV_CMD_REQUEST_VIDEO_STREAM_INFORMATION | WIP: Request video stream information (VIDEO_STREAM_INFORMATION)
  | Mission Param #1 | Camera ID (0 for all cameras, 1 for first, 2 for second, etc.)
  | Mission Param #2 | 0: No Action 1: Request video stream information
  | Mission Param #3 | Reserved (all remaining params)

Why do we have **0: No Action** in all the above MAVLink commands ?
For eg:

521 | MAV_CMD_REQUEST_CAMERA_INFORMATION | Request camera information (CAMERA_INFORMATION).
-- | -- | --
  | Mission Param #1 | 0: No action 1: Request camera capabilities
  | Mission Param #2 | Reserved (all remaining params)

What does it mean to pass Param 1 value as 0 ?
Passing value 1 (Request camera capabilities) should be implicit until we add a command for getting a specific camera information.
So, it is better to remove **No Action** field value from all these commands.

Please correct me if I am wrong or missed any information.

Thanks.

hamishwillee, 2018-04-12:
Yes, it doesn't make much sense. IMO all parameters should be reserved/ignored.

hamishwillee, 2019-02-06:
Closing. As discussed, this is a good idea, but not worth the cost of broken compatibility. See https://github.com/mavlink/mavlink/pull/887

closed 2019-02-06 by hamishwillee