mavlink/mavlink #888: Weekly MAVLink call and coordination

LorenzMeier, 2018-04-15:
Hi All,

Given the increased interest in standardization and MAVLink picking up more and more "micro-services" like the camera control standard, I believe a regular call would help. This is in particular necessary to coordinate efforts across different open source projects or vendors.

I would also like to facilitate the collaboration between different teams / groups and find common ground focused around technical merit. Calls and in-person meetings are essential to create the cohesion required for this.

I would propose a bi-weekly call, alternating between 09:00 a.m. CET / 15:00h Beijing / 17:00h Sydney time and 18:00 p.m. CET / 9 a.m. PST every 2nd Wednesday of the month.

I would also like to find a time for interested parties to meet at AUVSI Xponential early May in person.

Please leave feedback / suggestions in this thread.

ashirsat, 2018-04-15:
@LorenzMeier it would also be a good idea to have some sort of minutes or summary for the meeting that can be posted online so that if people miss the meeting they can get the updates

hamishwillee, 2018-04-16:
@amilcarlucas Would be great to have your involvement.

amilcarlucas, 2018-04-16:
Yes, I would like to participate

WickedShell, 2018-04-16:
I should be able to make either of the call times, and I will be at AUVSI as well.

keenanjohnson, 2018-04-17:
I would also be very glad to join the conversation and I'll also be around during AUVSI.

auturgy, 2018-04-17:
I won't be at AUVSI but am glad to see this effort.

LorenzMeier, 2018-04-19:
Thanks all for the feedback! I think I left enough time to comment for anybody interested initially - any comments on the times I proposed? Are they sensible or do we need something like a Doodle poll?

TSC21, 2018-04-19:
I think a Doodle would be a good option yes.

mrpollo, 2018-04-20:
Please join us for our first call on Wednesday, April 25th 2018, more info here https://mavlink.io/en/contributing/contributing.html#support

**Everyone is welcome!**

auturgy, 2018-04-20:
I think the first meeting needs an agreed agenda prior to scheduling: developing the framework and ground rules seems to me to be critical to success, and needs to be done early. Arbitrarily determining a time without determining where the key players are is not a good start.

mrpollo, 2018-04-20:
Thanks for the feedback @auturgy, we are trying to find the "key players," if this time doesn't work for you or others, please tell us what does work, and we will work it out as a group. I agree on the agenda; I'll post back with a link to contribute.

Everyone, this meeting should be seen as a "first step," we aren't claiming this first meeting is going to solve every problem for everyone, it will take **time, effort and coordination** to get this project where we all want it to be, let us get there together.

WickedShell, 2018-04-23:
I'm confused by what happened to the scheduling of this call? It was originally proposed as a biweekly call, (or every second Wednesday which would make it a monthly call) with alternating times to help other timezones. A weekly call seems like to be more frequent then is needed, but I'm mostly looking to clarify where the timing changed/target frequency.

keenanjohnson, 2018-04-24:
Is there going to be a meeting at AUVSI?

auturgy, 2018-04-25:
So it seems this meeting occurred without an agenda being sent out, or any real attempt to put this in the view of the largest mavlink users.
Can I ask that when this is rescheduled, the agenda be distributed prior, and notifications be sent not only to Slack, but gitter and discuss.ardupilot.org

LorenzMeier, 2018-04-29:
There was some confusion on the call - I specifically wanted to set up a Doodle so that we can find times that are non-discriminative. Consequently the call Ramon proposed was not an official MAVLink dev call and specs / direction were not discussed.

Here is the proper poll for the first official call:
https://doodle.com/poll/qmmprzpnr6z2hib4

Goal is to start in two weeks and then keep a weekly schedule on two time slots.

LorenzMeier, 2018-04-29:
In terms of AUVSI XPONENTIAL Meetup: Here is another Doodle Poll for this:

https://doodle.com/poll/2gmtxcipv2ghhkup

jacobsenmd, 2018-05-01:
Any chance we can get this nailed down? Thanks!

magicrub, 2018-05-01:
Time is Tuesday, May 1st from 2:00pm to 3:30pm.

LorenzMeier, 2018-05-01:
Just to make sure there is no confusion on location: Lobby of the Hyatt near the conference center (straight ahead of the main entrance)

mrpollo, 2018-05-07:
To everyone who attended, can we please get a summary of what was discussed and the outcome? Would love to hear how the first meeting went.

amilcarlucas, 2018-05-16:
When will the the meeting take place ??

LorenzMeier, 2018-05-16:
I just returned from travel, sorry about the delay on commenting here. I will provide feedback on the in-person meeting and coordinate the Doodle results with you.

magicrub, 2018-05-16:
Thanks @LorenzMeier

LorenzMeier, 2018-05-26:
Wednesdays 6-7 p.m Zurich / 9-10 a.m. PST seems to fit for most participants, so I would suggest starting with that on June 6, 2018.

Link to poll again: https://doodle.com/poll/qmmprzpnr6z2hib4

Meeting notes:

We had five attendees total from the APM and PX4 dev teams and discussed:

 - Coordinating some low-hanging fruits:
  - Well-defined behaviour of home position
  - Well-defined behaviour of MAVLink stream rates
 - Syncing the required fixes upstream and then getting rid of the pymavlink fork: https://github.com/mavlink/pymavlink
 - Setting up a weekly call to coordinate - the conclusion was that all interested participants are either based in Europe or the US
 - Working towards setting up a test suite to make it easier for implementers to comply with the spec
 - Establishing minimal base profiles that clearly define which messages and which micro-services have to be implemented
 - Auditing the language support such that we have clearly supported language bindings and "non-supported" ones that might not be as stable.

WickedShell, 2018-05-28:
I will not be able to make meetings at that time, but am interested in the outcome and look forward to the result.

auturgy, 2018-05-28:
Likewise (2am for me).

Regards,

James

hamishwillee, 2018-05-28:
Would it be beneficial/could the meeting be held at another time on a regular but reduced rate to pick up some of the other interested parties? E.g. one week in 4.  Just an idea.

OXINARF, 2018-05-29:
From the beginning it has always been the idea to have two alternating times because of the worldwide reach of MAVLink; I'm not sure how, for two times, we've ended up with a single meeting time.

auturgy, 2018-05-29:
> @OXINARF we followed the Doodle Poll that @LorenzMeier posted https://doodle.com/poll/qmmprzpnr6z2hib4#table
>
> Link to meeting invite for a weekly MAVLink call at 4:00 pm UTC https://calendar.google.com/event?action=TEMPLATE&tmeid=MGNxaDJybXBzb284bWF2MG9jNGE5bTQwdnJfMjAxODA2MDZUMTYwMDAwWiBsaW51eGZvdW5kYXRpb24ub3JnX2cyMXR2YW0yNG03cG03amhldjAxYnZscWg4QGc&tmsrc=linuxfoundation.org_g21tvam24m7pm7jhev01bvlqh8%40group.calendar.google.com&scp=ALL
>
> About the second call: I think having 2 calls that cover most timezones would be ideal, we just need to keep vigilant both groups coordinate since this can lead to further division, here's a poll to choose the second call time https://doodle.com/poll/hbgvdccezsmunnkf

The first poll was pretty clear about selecting two time slots. I’m not sure how that got confused.

Regards,

James

mrpollo, 2018-05-29:
There's not a clear winner that I can see from the remainder of timeslots for the first poll, the only one that caught my attention was the Tuesday @ 7:00 am UTC with 4 participants, I'm ok with scheduling whatever the community decides, I just don't want to guess.

hamishwillee, 2018-05-30:
@mrpollo Looking at the original poll, @auturgy , Michael and Thomas prefer 5-6pm Australia time. The day follows pattern Mon/Tues/Weds in subsequent weeks - I expect that this was somewhat random and they could probably all do any of those days - correct?

So perhaps pick in that timeslot and find a day that someone from the first group is also able to attend (to aid co-ordination)?

auturgy, 2018-05-30:
Thanks.
Can we kill the second poll? The way it is, likely to create more noise than signal.
And don’t forget to pick a time yourself Hamish!

hamishwillee, 2018-05-30:
I can do with 5-6pm on any day except (currently) Friday, and I'll fit in with what helps the team. 

I don't want to hijack the discussion until we've resolved a date. But once that is done I'm keen to better understand how "we keep vigilant both groups coordinate" (thanks Ramon). Specifically the mechanisms for coordinating issues that need more than just one group to agree.

LorenzMeier, 2018-05-31:
How about 5-6 p.m. Wednesdays, initially weekly, potentially bi-weekly? I've blocked that for now in my calendar so if its ok I could start pretty much any time.

hamishwillee, 2018-05-31:
> How about 5-6 p.m. Wednesdays, initially weekly, potentially bi-weekly? I've blocked that for now in my calendar so if its ok I could start pretty much any time.

Works for me! Even better, it means that key members of the community will be present in both meetings.  Ramon indicated he would try to attend in this timezone too.

auturgy, 2018-05-31:
Works for me too

Regards,

James

WickedShell, 2018-05-31:
What timezone is that 5-6PM in?

BTW the post here: https://github.com/mavlink/mavlink/issues/888#issuecomment-392868893 states 4PM UTC at the moment, the calendar link states 3PM UTC.

hamishwillee, 2018-05-31:
> What timezone is that 5-6PM in?

AEST - Melbourne/Sydney time

dagar, 2018-06-06:
Call in progress - https://zoom.us/meeting/579545140/ics?icsToken=fb96681cf4458be9f1f85694037a3cc7f358d5a9479ec0a06dedc267e49649ac

magicrub, 2018-06-06:
Aww crap. I missed the call and can't do the next time zone one. Next week is better. I talked to Michael dB and he says next week is better too. Sorry!

auturgy, 2018-06-06:
> @auturgy @WickedShell @hamishwillee here's the link to the calendar invite for Call 2 https://calendar.google.com/event?action=TEMPLATE&tmeid=MWxpY2x0NThzb3NvZGU5NmNxNzRqZm4xN2FfMjAxODA2MTNUMDcwMDAwWiBsaW51eGZvdW5kYXRpb24ub3JnX2cyMXR2YW0yNG03cG03amhldjAxYnZscWg4QGc&tmsrc=linuxfoundation.org_g21tvam24m7pm7jhev01bvlqh8%40group.calendar.google.com&scp=ALL
>
> Call time is 7:00am UTC, 5:00pm AEST. Starts next week (Wednesday 13)

Thanks, appreciated.

Regards,

James

hamishwillee, 2018-06-13:
Hi Everyone, just attended the MAVLink meeting 5pm AEST - no one but me and Lorenz there for most of it. We took the opportunity to discuss documentation: https://px4.slack.com/archives/C52DXKYLV/p1528875107000403

WOuld have been good if we could close out some open issues on this system. Was there something wrong with the notification?

auturgy, 2018-06-13:
Nothing wrong with the notification on your end afaict.
Seems I had a calendar sync issue at mine though.
Can help with the docs if you like.

Regards,

James

WickedShell, 2018-06-13:
I suffered a badly delayed flight and was driving home (without reliable phone service) through the call. Should not be a future issue/ongoing issue.

magicrub, 2018-06-13:
uhm, I added the two meetings from @mrpollo and none just occured.

This is getting ridiculous

magicrub, 2018-06-13:
Topics from the latest meeting:
- pymavlink fork needs to be solved, deferred to next meeting
- using common.xml as without a dialect. (opposite model of ArduPilot)
- Start thinking about MAVLink v3 to become the protocol to rule them all (long term goal)
- create a CI test that all autopilots and GCSs must test against to be compliant and to ensure they stay that way in the future.
- MAVLink project, independent of any autopilot, must dictate stricter usage rules (example: describe how to set home or mission sync).
- how to solve the old-msg-trash. Version. Deprecation policy.
- Extensions are broken because it allows adding extra ones and you don't know if they're populated or not (always zero). We need a better trash collection scheme.
- WIPs should not be allowed: Proposed solution is to move any new ones to special file that is not normally included such as WIP/Experimental/RFC.xml file
- get stakeholders to attend.

amilcarlucas, 2018-06-19:
Can someone please post the meeting times in UTC?
There is one in the evening and one in the morning right?

mrpollo, 2018-06-19:
Call times in UTC

Call 1: 7:00am UTC
Call 2: 4:00pm UTC

You can see the calendar and subscribe to any meetings here http://dev.px4.io/en/contribute/#calendar

mrpollo, 2018-06-19:
Just a heads up, the mavlink call is bi-weekly, we don't have a call scheduled this week (W25).

hamishwillee, 2018-06-27:
MAVLink meetings are scheduled for today. I plan on attending the one at 5pm AEST for first half hour (got another meeting at 5.30). 
- Who else is going?
- Does anyone have any specific issues for the agenda? What I would like to discuss is:
  - We had a few new issues/suggestions that did not get responses this week. Would be good to get some eyes on them - e.g.  https://github.com/mavlink/mavlink/issues/934
  - More discussion on things like versioning for which we had a flurry of activity but have not progressed.

auturgy, 2018-06-27:
I’ll be on the 5pm AEST call 

Regards,

James

hamishwillee, 2018-06-27:
MAVLink meeting Australia. I had to duck out early, but we covered:
1. [RFC9](https://github.com/mavlink/rfcs/pull/9)
   - not sure why system time can't be used. That may not always be correct, but will always be consistent for the vehicle in terms of linear time, across reboots.
   - need to understand whether this can be added as MAVLink2 (prefer not to add in MAVLink1)
1. Java Parser
   - Current implementation requires changes that will mandate breaking changes. 
   - Suggestions of changes have been met with silence. Impossible to progress without community interaction.
    - not clear if right people are seeing the discussions
    - stakeholders not clear - who is using parser, who can approve changes, who can contribute to parser
    - General feeling seemed to be that making breaking changes needed, so we need to identify stakeholders who can contribute/agree, then work on timing, notifications etc.
   - ACTION: @thomasgubler To take to next meeting to see if we have any interested parties.
1. We need process for sharing discussions across the two MAVLink meetings and between weeks.
    - General support for having a clear agenda and a scratchpad.
    - **Action** @mrpollo to set up google doc(s) for these in the Dronecode google drive. Share edit link with team for the scratchpad. [Don't you love getting agenda items from meetings you don't attend]

amilcarlucas, 2018-06-27:
Sorry could not attend. Will have to wait two more weeks :(

mrpollo, 2018-06-27:
@hamishwillee I think google docs is too restrictive for sharing minutes, my proposal is to use the [discourse forum](http://discuss.px4.io/c/mavlink)

2nd MAVLink call had some problems, Slack and Zoom where down for a while this morning, Liam, Thomas and myself where on the call.

auturgy, 2018-06-27:
I think google doc is better, and more bipartisan.
People shouldn’t need a login to your user forum to participate.

Regards,

James

mrpollo, 2018-06-27:
I have no problem with that @auturgy, but whoever is taking minutes on each call should create the doc and share access to the group, we still need to post those minutes somewhere for distribution and archiving.

dagar, 2018-06-27:
I don't feel that strongly about meeting minutes, but generally google docs isn't great for visibility. Over time the links end up lost or hard to find. If you post the content somewhere like discuss it ends up indexed by google.

I don't really see the problem with Dronecode discuss (you can login via github), but if it's not acceptable then I wouldn't be opposed to just dumping the notes straight to a github issue.

thomasgubler, 2018-06-28:
> I don't really see the problem with Dronecode discuss (you can login via github), but if it's not acceptable then I wouldn't be opposed to just dumping the notes straight to a github issue.

https://github.com/mavlink/mavlink/wiki  ?

mrpollo, 2018-06-28:
Whatever the maintainer team feels most comfortable with is fine with me, I don't have any preference, GitHub wiki works. To summarize the proposal is to use google docs and host the links on mavlink/mavlink wiki. Is anyone else in favor of this?  (or against).

hamishwillee, 2018-06-28:
Sounds good to me. 
- mavlink wiki for the agenda, linked to google docs for sharing discussion between the two meetings
- Might be nice to have the minutes summarised after the agenda, but that would depend on someone being happy to do that work.

[7 more replies]